realsenseai/librealsense #4776: Can there be constant values in variables when using one specific RGB and Depth camera?

kgh475926, 2019-09-01:
Camera Model: SR300
OS: Ubuntu
Platform: PC
Language: C++

I'm only using the camera SR300 for some project and I want to optimize some code to decrease the execution runtime. So I was just wondering if there can be constant values in variables when using one specific RGB and Depth camera (for this case, SR300)?

What I plan to do is to have all the constant variables and eliminate unnecessary calculations to speed up the execution process.

MartyG-RealSense, 2019-09-01:
I'm not sure how much processing you would save, but you can certainly turn off auto-exposure and use set values for variables such as exposure and gain.  You could also turn off AE Priority to enforce a constant FPS.

ev-mp, 2019-09-02:
@kgh475926 , the scope of available optimizations is  use-case tailored so if you can describe your use-case/platform then we would be able to assess which optimizations may benefit the code (e.g. CUDA/OpenMP).
In the SDK we also have a built-in utility class named `scoped_timer` that measures function execution times, so you can use it to profile `librealsense` code and to highlight the time-consuming routines.

Certain generic parts, such as YUYV->RGB color-space conversions, already have an optimized version in the core library. Additionally SDK provides several optimizations that take advantage of stream-predefined values. For instance, in the SSE implementation of Alignment processing block chunks of de-projection data that relies solely on the camera intrinsic matrix are precomputed for efficiency.

kgh475926, 2019-09-02:
@ev-mp You are actually at what I'm trying to get. I'm trying to figure out a way to optimize the function move_depth_to_other in sse-align.cpp because I want to optimize the whole process of alignment. You have said that the camera intrinsic matrix are precomputed for efficiency but do you think the code implementation of the function move_depth_to_other can be optimized further or would you say it has been optimized to the fullest?

ev-mp, 2019-09-03:
@kgh475926, you can definitely profile the code and propose  additional optimizations both on micro and macro levels.
I'm hesitant to make suggestions as it is often platform/OS-specific. But if you're running a modern X64 CPUs then one of the low-risk optimization would be porting SSE to AVX.

kgh475926, 2019-09-03:
@ev-mp I see. Thank you for the suggestions I really appreciate it.

closed 2019-09-18 by ev-mp