realsenseai/librealsense #4777: A question about 360 degree captures

liltrekkie, 2019-09-02:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | Latest | 
| Operating System & Version | Win 10 | 
| Platform                                 | PC |
| SDK Version                            |  2.27 | 

### Issue Description

So I have a couple questions about doing 360 degree captures, there was another issue that talked about this some but didn't exactly go into a whole lot of detail, the varying consensus was you either do multi cam, or use a turn table, now for our use case, the turn table is the best option, now here are my questions.

Question 1: When using the realsense viewer, when you export a 3d point cloud, it appears to just be a snap shot of a given frame, is there a way to export an entire recording as a point cloud? If there isn't, then this leads to question 2,. 

Question 2: Would I be able to do this through the API? This would probably be more convenient in the long run anyway. If I can't, this leads to question 3.

Question 3: If by some chance I am unable to get an entire 360 point cloud in one shot, is there a way whether it be through the viewer or the API where I could create a bunch of snapshots say every second or half second and then combine them?

liltrekkie, 2019-09-09:
It seems there is a limitation on the timestamps I can get 3d snapshots from the viewer, so I can only get a single one every second, which isn't enough to combine the point clouds together successfully. 

So would anyone have any other recommendations?

RealSenseCustomerSupport, 2019-10-01:
----------------------------------------------

Similar topic was discussed in the following thread. The well organized and thoughtful information provided by MartyG-RealSense are extremely helpful on combining point-clouds.
https://github.com/IntelRealSense/librealsense/issues/3496

liltrekkie, 2019-10-06:
Thank you! I had searched through all the issues pertaining to what I was looking for, but somehow I missed that one. It contains a lot of useful information. 

Based on the information provided by the various issues, and specifically that one, there doesn't seem to be a way to capture a single point cloud. Which answers most of my questions. 

However, the realsense SDK provides a converter that converts to a .ply, which is probably the route we will go down, and then we will stitch the point clouds together. It doesnt seem to output anything for me though. All the other options for the converter does work. Should I create a separate issue for this?

RealSenseCustomerSupport, 2019-10-07:
----------------------------------------------

Thank you for the feedback and I am glad to hear the link provided was useful.
As for rs-convert tool, please try to record both depth and rgb streams into a bag file and see if the problem persists.

liltrekkie, 2019-10-07:
I have, I've tried with and without RGB streams, with and without motion, I've tried reinstalling the latest version, wiping and reinstalling, I've tried multiple different bag files, including all of those in the sample data section. It doesnt matter how long or short the bag file is either. It just won't convert to ply no matter what I try.

RealSenseCustomerSupport, 2019-10-08:
----------------------------------------------

Hi, it's a good idea to try the sample data and I  double checked one of the recording samples for D435i with the rs-convert built from SDK 2.29 and it seem worked as expected on my systems (Ubuntu 18.04 and Win10).
Thank you.

    :~/Downloads/d435i_sample_data$ ~/Work/librealsense-2.29.0/build/tools/convert/rs-convert -i ./d435i_walking.bag -l ./ply/99%
    PLY converter
     258  frame(s) processed

    :~/Downloads/d435i_sample_data$ ll -R  | more.:
    total 1528088
          4096 Oct  8 14:50 ./
          4096 Oct  8 14:56 ../
          224 Oct  8 15:08 2019-10-08-14_46_08.log
     841910563 Nov 25  2018 d435i_walk_around.bag
     722813137 Nov 25  2018 d435i_walking.bag
        12288 Oct  8 15:08 ply/

    ./ply:
    total 6808792
        12288 Oct  8 15:08 ./
        4096 Oct  8 14:50 ../
     26186727 Oct  8 14:46 _1503.ply
     26347719 Oct  8 14:46 _1504.ply
     26626334 Oct  8 14:46 _1505.ply
     26426509 Oct  8 14:46 _1506.ply
     26585203 Oct  8 14:46 _1507.ply
     26637192 Oct  8 14:46 _1508.ply
     26536221 Oct  8 14:46 _1509.ply
     26726347 Oct  8 14:46 _1510.ply
     26709393 Oct  8 14:46 _1511.ply
     26684960 Oct  8 14:47 _1512.ply
    --More--

liltrekkie, 2019-10-09:
Maybe I am doing something wrong then. Here is the command I use 

`rs-convert.exe -i "d435i_walk_around.bag" -l output`

Now after I use that, the cursor blinks for about 5 seconds, then command line reappears and waits for my next command. Both the .bag file, and the converter are in the same folder, as the instructions state. and I just updated to the latest build. Am I just by chance entering the command wrong or something?

RealSenseCustomerSupport, 2019-10-10:
----------------------------------------------

Still cannot reproduce the issue you have and the command you provided also worked here.
Any error or warning message in the *.log file?
Thank you.

liltrekkie, 2019-10-10:
It creates a log file, but it is blank. I can literally use ever other possible converter with success, except ply. Ply is the only converter that doesn't work for me.

liltrekkie, 2019-10-16:
[image]

This is a screenshot of the powershell of what happens when i execute the script. It does the same thing in command prompt too. Both of these are with admin privileges. As you see, there is no output, nothing. And the log is blank too. This is using the latest posted version, from a complete uninstall and reinstall. Is there anything I might be doing wrong that would cause this to happen?

RealSenseCustomerSupport, 2019-10-18:
----------------------------------------------

I think you are doing it correctly but it is strange why only the PLY doesn't work on your side.
By any chance you can try it on another Windows 10 PC so we know this problem isn't isolated only in one particular system setup?
Thank you.

Attachment(s):
rs-convert-windows.png - https://realsensesupport.intel.com/attachments/token/yXg7T019wZXG3zJdtbT3kSQwv/?name=rs-convert-windows.png

liltrekkie, 2019-10-20:
I've tried it with a handful of PC's now. I've even tried it on VM's, with no success. I had a friend try it as well, with zero success.

RealSenseCustomerSupport, 2019-10-24:
----------------------------------------------

Just for debugging purposes, does this work?
RealSense Viewer > Add Source > Load Recorded Sequence > Switch to 3D view > Click Export 3D model to PLY format

liltrekkie, 2019-10-27:
Yes, I can export ply's from the realsense viewer. It's just the converter that does not work.

RealSenseCustomerSupport, 2019-11-08:
----------------------------------------------

Just curious if by any chance you have found the solution or w/a to convert bag into ply.
We are still not able to reproduce this issue on our side.

liltrekkie, 2019-11-08:
I actually have made some progress on this. 

I moved the convert tool out of its folder, after which it refused to work at all then, which makes sense, I moved it back in the tools folder, how when I try to use the sample data, nothing happens. However, when I use my own bag file, It attempts to convert it, however, despite it saying 100%, nothing converts, and when you open the file, you get both a warning, and an error. 

` 08/11 15:04:43,360 WARNING [3908] (playback_device.cpp:198) Playback device does not provide a matcher
 08/11 15:04:46,841 ERROR [10988] (playback_device.cpp:474) Failed to read next frame from file: Required '' field missing`

I have used all the other sample data available, and it yields the same result, I have all used every bag files I've ever created, a total of 10, and they all create the same error. I have been able to recreate this error just by dragging the tool out of the folder, and putting it back in. 

At least it tries to do something now, at least. Aha.

liltrekkie, 2019-11-13:
A small update, I went out and built an Intel based system (9900k) and a zen based system (3900x) and installed realsense. The ply converter still does not work on them. I'm not sure what else I can even do at this point. Is there any debugging that I can do that would be useful here?

RealSenseCustomerSupport, 2019-11-14:
----------------------------------------------

Build and debug from rs-convert source may be a good way to narrow down where the issue is.
https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_windows.md

Also see the attachment for a very short bag file and the rs-convert tool with more verbose log.

Attachment(s):
rs-convert.zip - https://realsensesupport.intel.com/attachments/token/BoJJolLfsqXcdoH34aktVtkYj/?name=rs-convert.zip

RealSenseCustomerSupport, 2019-11-14:
----------------------------------------------

Please try the following link if you have any permission issue from the previous one. Thank you.

https://intel.sendsafely.com/receive/?thread=J184-LRMG&packageCode=C0tdE8xtIHzpgEcMn7WfVhFht80mt0VCTUkU5Cgjfls#keyCode=sK5EjNOLgBsCNo03eIPvdfUIweHnqYeDKUuWNnyQJd0

liltrekkie, 2019-11-14:
[2019-11-14-02_13_51.log](https://github.com/IntelRealSense/librealsense/files/3845582/2019-11-14-02_13_51.log)

I ran what you gave me, and this is the file that is produced, as for building from source, I have tried that too, which it didn't work either. I tried building it from source on linux as well, again, nothing. If I run what you gave me on linux, a file with similar contents is created. If it helps I could upload it too. Hopefully this file sheds some light on what may be happening.

RealSenseCustomerSupport, 2020-01-07:
----------------------------------------------

just curious if there is any update on this issue? Thank you.

liltrekkie, 2020-01-07:
There is not. Nothing I've tried has worked. This is the most infuriating experience ever. Multiple sets of hardware have been tried, none have any success. I've tried multiple bag files. Even bought another realsense camera to see if that was somehow the issue. It wasn't. 

Everything except ply works. I've tried tinkering with every setting in the viewer app there is. Defaults, turning everything off, searching through issues to see if there is any fix what so ever. Nothing has worked. 

It is upsetting, to say the least.

RealSenseCustomerSupport, 2020-01-08:
----------------------------------------------

@liltrekkie, thank you for the update and I am sorry to hear the issue still remains.
I understand your frustration and I also tested the sample bag file on several systems I have (i5 or i7 with Ubuntu 18.04 or Win10); however, the issue has not yet been reproduced here.
By the way,  by any chance that you are already NDA customer with Intel and has a Zendesk account?

liltrekkie, 2020-01-08:
I do not have a zendesk account, I could create one pretty easily though, but I'm not an Intel NDA customer either, although I have zero issues becoming one. 

I do have a small update, I was curious and tried this on a Windows Server 2012 R2 VM, using the windows 7 install, since 2012 is based on windows 8 and I didn't think the windows 10 one would work, and used the stairs.bag sample data, and it exports the ply's successfully! Which is a massive jump. However, the windows 10 realsense version one still doesn't work on a windows 10 VM or using Windows Server 2016.

RealSenseCustomerSupport, 2020-01-29:
----------------------------------------------

Let's close this ticket since the original question was answered. However, please feel to file another ticket here or on the Zendesk if further assistance is needed.
Thank you.

closed 2020-02-04 by RealSenseCustomerSupport