realsenseai/librealsense #4778: Snap [Feature Request]

mikeroyal, 2019-09-02:
I was thinking the install and setup could be improved for Ubuntu and other Linux distros by having a [snap](https://snapcraft.io/) version. If it's not a priority would you mind if I made the snap as a contribution?

MartyG-RealSense, 2019-09-02:
Intel welcome community contributions of any kind for RealSense.  In an online 'webinar' session about RealSense in June 2018, the following statement was made by the Intel presenters:

"We strongly appreciate and encourage community participation ... we wanted to make it very easy for people to take the library and modify it to their needs, and share it with more people, even if it doesn't line up with our vision of the product".

If a customer writes a feature, documents and submits it, then it is "a very strong tool for influencing the direction of the product".  

Intel will almost always happily accept a contribution and try to maintain it from that point onward.

dorodnic, 2019-09-10:
I support this feature request.
Currently the SDK has dkms and udev-rules packages that may be tricky to package. DKMS can be bypassed by building the SDK with `-DFORCE_LIBUVC=true`, but udev-rules are a must. I'm sure there are solutions for it, but I personally am not familiar with Snap at all.

MartyG-RealSense, 2019-09-10:
@dorodnic  Snaps were a new concept to me too.  From what I can gather from research, "Snaps are applications packaged with all their dependencies to run on all popular Linux distributions from a single build.  They update automatically and roll back gracefully".

A couple of years ago, someone released a Snap for Legacy Librealsense (1.12).

https://snapcraft.io/librealsense

It seems that in that system, Librealsense could be installed via Snap using a pip-style command:

**sudo snap install librealsense**

mikeroyal, 2019-09-22:
Hi @kyrofa,

I wondering if you could help me out here I was wanting to make an update version of the librealsense SDK snap but I noticed Canonical made one back in May 2017. Do you know who made the original one? Or someone who could update the snap to the upcoming [2.28](https://github.com/IntelRealSense/librealsense/releases/tag/v2.28.0) build.

kyrofa, 2019-09-22:
@mikeroyal thanks for the ping. I've passed it onto the folks responsible and asked them to participate in this discussion.

RealSenseCustomerSupport, 2020-01-14:
----------------------------------------------

Hi @mikeroyal, Do you still plan to submit a PR for Snap? Thanks.

RealSenseCustomerSupport, 2020-01-24:
----------------------------------------------

Hi @mikeroyal, Do you still plan to submit a PR for Snap? Thanks.

**Please be informed that we will close this ticket in seven (7) days if we have not heard back from you.

closed 2020-01-31 by RealSenseCustomerSupport