realsenseai/librealsense #4779: Malformed error

Maheshwariraghav112, 2019-09-02:
E: Malformed entry 1 in list file /etc/apt/sources.list.d/realsense-public.list (Component)

ev-mp, 2019-09-02:
@Maheshwariraghav112 , this file `/etc/apt/sources.list.d/realsense-public.list` is not used by Librealsense and has been replaced by alternative and [streamlined approach](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#installing-the-packages) a while ago.  
It is still mentioned in `intel-realsense-dfu` [installation guide](https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/Linux-RealSense-D400-DFU-Guide.pdf) though.

If you use `intel-realsense-dfu` package then I'd recommend to delete and make a new version of the file. Otherwise - deleting this file should be safe.

Maheshwariraghav112, 2019-09-03:
Thanks for your response . 
As you told i already deleted all the packages but still having the same problem .

ev-mp, 2019-09-03:
@Maheshwariraghav112, I think I didn't express myself clear - There is no need to uninstall any Debian packages.
Just delete the mentioned file and run `sudo apt-get update` to update the cache.
In case of using `intel-realsense-dfu` create a new version of that file according to the installation guide.

ev-mp, 2019-09-18:
@Maheshwariraghav112 , did you manage to resolve the issue ?

ev-mp, 2019-09-26:
@Maheshwariraghav112 , feel free to reopen as needed.

closed 2019-09-26 by ev-mp