realsenseai/librealsense #4780: install librealsense2-dkms error

comwise, 2019-09-02:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435i                                                     |  
| Firmware Version                   |  | 
| Operating System & Version |  Ubuntu 16.04                                        | 
| Kernel Version (Linux Only)    |   4.14.0-31                                      | 
| Platform                                 |  intel PC  |
| SDK Version                           |  2.28                           | 
| Language                               |  {C++/python                         | 
| Segment			         |  Robot                           | 

### Issue Description
<Describe your issue / question / feature request / etc..>
installl librealsense2-dkms package report the following error, other package is ok
i check the linux header file, i found the marco UVC_PC_PROTOCOL_15 which is defined in file
/usr/include/linux/usb/video.h not in "include/linux/usb.h",
--------
正在设置 librealsense2-dkms (1.3.5-0ubuntu1) ...
Loading new librealsense2-dkms-1.3.5 DKMS files...
Building only for 4.4.0-31-generic
Building initial module for 4.4.0-31-generic
Error! Bad return status for module build on kernel: 4.4.0-31-generic (x86_64)
Consult /var/lib/dkms/librealsense2-dkms/1.3.5/build/make.log for more information.
--------
make: Entering directory '/usr/src/linux-headers-4.4.0-31-generic'
  CC [M]  /var/lib/dkms/librealsense2-dkms/1.3.5/build/4.4.0/drivers/media/usb/uvc/uvc_driver.o
In file included from /var/lib/dkms/librealsense2-dkms/1.3.5/build/4.4.0/drivers/media/usb/uvc/uvc_driver.c:19:0:
/var/lib/dkms/librealsense2-dkms/1.3.5/build/4.4.0/drivers/media/usb/uvc/uvc_driver.c:2990:43: error: ‘UVC_PC_PROTOCOL_15’ undeclared here (not in a function)
  { USB_INTERFACE_INFO(USB_CLASS_VIDEO, 1, UVC_PC_PROTOCOL_15) },
                                           ^
include/linux/usb.h:952:25: note: in definition of macro ‘USB_INTERFACE_INFO’
  .bInterfaceProtocol = (pr)
                         ^
scripts/Makefile.build:258: recipe for target '/var/lib/dkms/librealsense2-dkms/1.3.5/build/4.4.0/drivers/media/usb/uvc/uvc_driver.o' failed
make[1]: *** [/var/lib/dkms/librealsense2-dkms/1.3.5/build/4.4.0/drivers/media/usb/uvc/uvc_driver.o] Error 1
Makefile:1573: recipe for target 'uvcvideo.ko' failed
make: *** [uvcvideo.ko] Error 2
make: Leaving directory '/usr/src/linux-headers-4.4.0-31-generic'
-------------

other one (4.4.0-159) is install from online  is ok

can you  tell me how to do this?

ev-mp, 2019-09-02:
@comwise , D435i is implemented according to UVC 1.5 standard and the kernel you're running is pre-UVC1.5 (UVC1.1). So even if you manage to resolve compilation issues I don't think it is guaranteed that the device (or at least some controls) will work properly.
Officially UVC1.5 support was added in kernel 4.5, but with most distros (e.g Ubuntu) any kernel beyond ~4.4.0-60+ should be ok, like the 4.4.0-159 that you mention.

Is there any specific reason that you have to use 4.4.0-31 and not the up-to-date 4.4.0-159?

comwise, 2019-09-03:
@ev-mp this linux image is clone，all old robots image is base on kernel 4.4.0-31,  if i upgrade the kernel，some devices is unusable，i need to update these device drivers which is a difficulty and complex job, if you can offer good advice, i will upgrade the linux kernel

ev-mp, 2019-09-03:
@comwise  Did you try to update the system? It should be really streamlined - 
>sudo apt-get update && sudo apt-get upgrade && sudo apt-dist-upgrade -y

The kernel 4.4.0-31 was built into the baseline image, and this apparently the baseline Ubuntu used when started shipping Xenial.
As I mentioned above that kernel is not compatible with D435i as it does not include driver support for UVC1.5 standard. So in case you want to install DKSM upgrading the kernel is necessary.
There are additional benefits for getting the system updated - bug and security fixes, such as the notorious Meltdown and Spectre. 

Note the DKMS is not a hard-requirement and in many case can be dismissed/replaced by `libuvc` backend. So this is another path you can follow if you're stuck with 4.4.0-31.

comwise, 2019-09-20:
@ev-mp  i updated the kernel by the following 
`sudo apt-get update && sudo apt-get upgrade`
which can resolve my driver problems, 

if i only updated kerner by 
` sudo apt-get install linux-image-4.4.0-159-generic linux-image-extra-4.4.0-159-generic,`  
there is some problems with drvier

thanks for your reply

RealSenseCustomerSupport, 2019-10-09:
----------------------------------------------

Hi @comwise,
Glad to hear the problem is resolved.
Closing this ticket unless there is any other question.

closed 2019-10-17 by RealSenseCustomerSupport