realsenseai/librealsense #4781: ros-kinetic-librealsense fails on install within container

dizz, 2019-09-02:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | n/a | 
| Firmware Version                   | n/a | 
| Operating System & Version |  Ubuntu 16.04 LTS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  legacy                          | 
| Language                            |  bash                         | 
| Segment			|  Robot                          | 

### Issue Description

**Versions:**
 * Host OS: `Ubuntu 16.04 LTS`
 * Platform: `docker`
 * OS: Ubuntu `16.04 LTS`
 * ros-kinetic-realsense: `ros-kinetic-librealsense_1.12.1`

Hi,

We’ve had an issue installing the `ros-kinetic-librealsense` package within our ROS related containers, where the `postinst` deb script fails with:

```
Setting up ros-kinetic-librealsense (1.12.1-0xenial-20190607-170900-0800) ...
debconf: unable to initialize frontend: Dialog
debconf: (Dialog frontend will not work on a dumb terminal, an emacs shell buffer, or without a controlling terminal.)
debconf: falling back to frontend: Readline
/var/lib/dpkg/info/ros-kinetic-librealsense.postinst: 40: /var/lib/dpkg/info/ros-kinetic-librealsense.postinst: udevadm: not found
libkmod: ERROR ../libkmod/libkmod.c:586 kmod_search_moddep: could not open moddep file '/lib/modules/4.15.0-1028-gcp/modules.dep.bin'
modinfo: ERROR: Module alias uvcvideo not found.
dpkg: error processing package ros-kinetic-librealsense (--configure):
 subprocess installed post-installation script returned error exit status 1
Setting up ros-kinetic-openni2-camera (0.4.2-0xenial-20190608-120620-0800) ...
Setting up ros-kinetic-openni2-launch (0.4.2-0xenial-20190608-132724-0800) ...
dpkg: dependency problems prevent configuration of ros-kinetic-realsense-camera:
 ros-kinetic-realsense-camera depends on ros-kinetic-librealsense; however:
  Package ros-kinetic-librealsense is not configured yet.
```

The issue comes from the deb `postinst` script check that sees if the package is installed in a container or without one. Without one it works as expected, however when ran from within a container it fails (the script check reports that it’s executed not from within a container). So the following command in `postinst` is buggy: `docker_cnt=$(awk '$2 == "(1,"' /proc/1/sched  | wc -l)`

**Fix**
What we have done is to update the postinst script such that it detects the presence of a container better (hopefully!) with the following command: `docker_cnt=$(awk -F/ '$2 == "docker"' /proc/1/cgroup  | wc -l)` and had to consequentially invert the conditional checks.

Those [fixes can be seen here in this gist](https://gist.githubusercontent.com/dizz/404ef259a15e1410d692792da0c27a47/raw/3769e80a051b5f2ce2a08d4ee6f79c766724f495/postinst). The lines of note are 29, 37 and 65

The steps we have taken to resolve the issue locally are:

```
# basic container setup
apt-get update
apt-get install ssh lsb-release
apt-key adv --keyserver 'hkp://keyserver.ubuntu.com:80' --recv-key C1CF6E31E6BADE8868B172B4F42ED6FBAB17C654
echo "deb http://packages.ros.org/ros/ubuntu $(lsb_release -sc) main" > /etc/apt/sources.list.d/ros-latest.list
apt-get update

# dependencies needed by librealsense. `deb -i` will not resolve these
apt-get install -y binutils cpp cpp-5 dkms fakeroot gcc gcc-5 kmod libasan2 libatomic1 libc-dev-bin libc6-dev libcc1-0 libcilkrts5 libfakeroot libgcc-5-dev libgmp10 libgomp1 libisl15 libitm1 liblsan0 libmpc3 libmpfr4 libmpx0 libquadmath0 libssl-dev libssl-doc libtsan0 libubsan0 libusb-1.0-0 libusb-1.0-0-dev libusb-1.0-doc linux-headers-4.4.0-159 linux-headers-4.4.0-159-generic linux-headers-generic linux-libc-dev make manpages manpages-dev menu patch zlib1g-dev
apt-get install -y libssl-dev libssl-doc libusb-1.0-0 libusb-1.0-0-dev libusb-1.0-doc linux-headers-4.4.0-159 linux-headers-4.4.0-159-generic linux-headers-generic zlib1g-dev

# modify librealsense deb (unpack, replace script, repack)
apt-get download ros-kinetic-librealsense
dpkg-deb -R ros-kinetic-librealsense*.deb ros-rslib/

wget https://gist.githubusercontent.com/dizz/404ef259a15e1410d692792da0c27a47/raw/3769e80a051b5f2ce2a08d4ee6f79c766724f495/postinst
chmod +x postinst
cp postinst ros-rslib/DEBIAN
dpkg-deb -b ./ros-rslib/ ros-kinetic-librealsense_1.12.1-0xenial-20190830_icrlab_amd64.deb

# install container friendly libsense
dpkg -i ros-kinetic-librealsense_1.12.1-0xenial-20190830_icrlab_amd64.deb

# lock from updates
sudo apt-mark hold ros-kinetic-librealsense
```

This then resolves the issues related to the install within a container.

Perhaps someone could point me to the person responsible for the builds so we could think about a patch?

Cheers,
Andy

PS: While there was a build issue on Ubuntu 16.04 LTS, the same problem did not show itself using a MacOS based docker build.

RealSenseCustomerSupport, 2019-12-05:
----------------------------------------------

@dizz
Since SDK 1.0 is not maintained, still thanks for your contribution.

closed 2019-12-05 by RealSenseCustomerSupport

yf-commit, 2020-06-18:
How to do you solve this problem? I have the same problem when I tried to install ros packages within the container.
Errors were encountered while processing:
 ros-kinetic-librealsense

dizz, 2020-06-18:
Manual instructions to fix are in this post. It's possible that they're now outdated, however.

lalten, 2021-12-19:
@doronhi's fix does not work for me ( `docker run osrf/ros:kinetic-desktop-full cat /proc/1/cgroup` returns `0::/`). This may be an Arch thing, or Kernel 5.15 :shrug: 

In any case, I added this fix to my Dockerfile to have the postinstall just always assume it's in a container.
```Dockerfile
RUN apt download ros-kinetic-librealsense && \
    dpkg-deb -R ros-kinetic-librealsense* ros-rslib && \
    sed -i 's/^docker_cnt=$(.*/docker_cnt=1/' ros-rslib/DEBIAN/postinst && \
    dpkg-deb -b ./ros-rslib/ ros-kinetic-librealsense-fixed.deb && \
    dpkg -i ros-kinetic-librealsense-fixed.deb ; \
    apt --fix-broken --yes install && \
    rm -rf ros-rslib ros-kinetic-librealsense-*
```
Hope this helps someone :)