realsenseai/librealsense #4782: missing using Intel.RealSense in U3D, who solved?  "The type or namespace name `Intel' could not be found. Are you missing an assembly reference?"

vangewj, 2019-09-02:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

dorodnic, 2019-09-03:
Have you tried https://github.com/IntelRealSense/librealsense/releases/download/v2.27.0/Intel.RealSense.unitypackage ?

vangewj, 2019-09-04:
support to capture face  78points  sdk of Unity ?use D435 RealSense

RealSenseCustomerSupport, 2019-09-25:
----------------------------------------------

Hi @vangewj,

About "capture face 78points", this is supported by the legacy SDK. We don't provide this middleware in SDK2.0. You can consult with our 3rd party software vendors.
https://www.intelrealsense.com/software-for-intel-realsense/

If no further issue for this ticket, I will close it.

closed 2019-10-16 by RealSenseCustomerSupport