realsenseai/librealsense #4783: Transform Pose from IR frame to Color Frame

lolanda1000, 2019-09-02:
Hello,
Is there an easy way to transform an pose calculated in the Infrared Stream to the Color Stream?

ev-mp, 2019-09-02:
@ghost hello,
 Librealsense does not cover direct 2D<=>2D alignment method, which are more in the CV domain.
This is feasible however, in Left IR->RGB scenario  by using the pointcloud class.

Pointcloud that receives RGB and Depth frames produces a "points" frame which comprise of vertices and texture [u,v] coordinates buffers.
So if you know the pixel coordinates of the area of interest in Left IR frame (Left IR and Depth are perfectly aligned) then you can access the [u,v] coordinates at the same indexes to retrieve the RGB pixels that correspond in the IR view.
Pseudo code based on `rs-pointcloud` example:
```
rs2::pointcloud pc;
auto frames = pipe.wait_for_frames();
auto color = frames.get_color_frame();
// Tell pointcloud object to map to this color frame
pc.map_to(color);
auto depth = frames.get_depth_frame();
// Generate the pointcloud and texture mappings
points = pc.calculate(depth);
auto tc = points.get_texture_coordinates();
ir_pixel = [123,456]; // Point of interest in the left IR frame
rgb_uv = tc[123*depth_width+456]; // Location of U,V coordinate that corresponds to the requested index
rgb_pixel = [rgb_uv ].u*rgb_height [rgb_uv ].v* rgb_width]; // * U,V are normalized
```
Two caveats here:
- The leftmost depth pixel columns are zeroed due to the stereo constrains, so you'll not find correspondence for IR pixels on the left edge.
- While the IR data is available for all pixels, it does not hold for the Depth frame, so for Depth pixels with no actual data you'll not be able to extract the RGB counterparts.

Performance-wise the overhead of this approach is that in addition to RGB and IR you also need to stream depth data, and then process and generate the points frame.

ev-mp, 2019-09-18:
@ghost, was this helpful ?

RealSenseCustomerSupport, 2019-10-09:
----------------------------------------------

@ghost, please let us know  if you need further help on this topic. Thank you.

closed 2019-10-17 by RealSenseCustomerSupport