realsenseai/librealsense #4789: T265 Pose Inaccuracy

pietroglyph, 2019-09-03:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   | 0.0.18.5715 | 
| Operating System & Version |   Linux  | 
| Kernel Version (Linux Only)    |  4.9.47                                         | 
| Platform                                 | NI RoboRIO  |
| SDK Version                            |  2.24.0                          | 
| Language                            |  C++                          | 
| Segment			|  Robot                          | 

### Issue Description
The T265 camera can sometimes have pose error along the Z or X axis of nearly 25 cm (10 inches). Other times, the error is as low as < 2.5 cm (1 inch).

I ran a number of experiments where I controlled for approximate velocity during travel, distance and path traveled, wheel odometry on/off, measurement covariance, and objects in the scene. The only thing that changed between a trial where I got lots of error and a trial where I got very little was the lighting in the room. It was during dusk (when there was less harsh light) that I had the least error. During the night (with lights on) there is error up to 13 cm (5 inches), and during certain times of the afternoon the error is at its highest (25 cm) and the light is also brightest.

To the human eye and other cameras the scene looks fine, but to the T265 the windows (even if lightly covered) are totally blown out except during certain times of day. Wheel odometry being on or off does not change accuracy meaningfully (it does make the camera resistant to total failure, like if my hand covers both sensors, so I think I'm sending the correct data). Setting the exposure mode to manual and tuning it didn't make a very large difference in accuracy either.

<hr/>

[Here is a link to a recording where I get about 25 cm of error,](https://drive.google.com/file/d/1SXYo5_MCuRrHtVylOMjcUXq4VUqpbc82/view?usp=sharing) and below is a frame from that recording:
[image]

Here is a picture of the scene from my phone:
[image]

<hr/>

With the above context, I have three questions:
 1. Is this amount of error "normal" for this specific scene and lighting?
 2. Is it plausible that this is mostly a lighting issue?
 2. Is there anything else I can do to mitigate this error other than changing the scene? Would a physical filter over the camera help? What about adding a feature to change auto exposure settings?

JimmyFroost, 2019-10-07:
hello @pietroglyph 

I am facing the same problem, T265 gives me a bad trajectory when I am facing to a large window with bright light coming through: resulting path is shorter than expected or the rotation is not right.

I am looking for "regular camera " solution, such as polarized filter, etc. but I don't think it will be enough.

Have you had new interesting results from experimenting on this light problem ?

pietroglyph, 2019-10-07:
@JimmyFroost Unfortunately I haven’t had any success with this issue. Perhaps someone from Intel could provide some support, especially because this supports more than one user.

JimmyFroost, 2019-10-08:
I agree, this should be adressed by Intel, maybe they could propose a "Sunshade Prevent Glare"  object for the camera.
@MartyG-RealSense , have you experimented on this issue ?

MartyG-RealSense, 2019-10-08:
@JimmyFroost   I have not got access to a T265, so I am not able to experiment with it, unfortunately.  There are certainly ways to apply physical filters (e.g micro-thin filter film) over lenses though to change the properties of light entering the imagers.

JimmyFroost, 2019-10-08:
Ok thank you @MartyG-RealSense  for your reply. Is there anyone you know that might help us ?

I quickly tried using sunglasses over T265 indeed, it can be a way to solve part of the problem? I will probably look more into it in the next couple of months...

MartyG-RealSense, 2019-10-08:
The link below collects together in a single message a number of approaches to using filters and 'cover materials'.  They are written in terms of the 400 Series depth cameras, but I would think that the principles could be applied to the T265 too.

https://forums.intel.com/s/question/0D70P0000065HxqSAE

JimmyFroost, 2019-10-08:
thanks again, I'll keep that in mind.

RealSenseCustomerSupport, 2019-10-08:
----------------------------------------------

Thanks @MartyG-RealSense for commenting regarding filters as a possible solution.  We have not done this type of extensive film/filter testing on T265 as was done on the depth cameras.  When it comes to the T265 and tracking using it, the environment and what is in it and lighting etc is very much an important part to the tracking algorithm and how it functions.  In very bright rooms, bright sun coming in from windows directly in the cameras view can wash out many of the feature points the T265 uses to determine location in the room.

closed 2019-10-31 by RealSenseCustomerSupport