realsenseai/librealsense #4793: Building with Cuda on windows

nawara72, 2019-09-03:
Hi

I am trying to build librealsense on windows with Visual studio. I have no issues if building with no cuda. However if i try to build with cuda (using cmake then visual studio) i get these errors

I would appreciate any help to resolve this as soon as possible

Thanks
Nawar

ds5-options.obj : error LNK2038: mismatch detected for 'RuntimeLibrary': value 'MTd_StaticDebug' doesn't match value 'MDd_DynamicDebug' in cuda-align.obj
1>ds5-timestamp.obj : error LNK2038: mismatch detected for 'RuntimeLibrary': value 'MTd_StaticDebug' doesn't match value 'MDd_DynamicDebug' in cuda-align.obj
1>ds5-private.obj : error LNK2038: mismatch detected for 'RuntimeLibrary': value 'MTd_StaticDebug' doesn't match value 'MDd_DynamicDebug' in cuda-align.obj
1>ds5-motion.obj : error LNK2038: mismatch detected for 'RuntimeLibrary': value 'MTd_StaticDebug' doesn't match value 'MDd_DynamicDebug' in cuda-align.obj
1>ds5-nonmonochrome.obj : error LNK2038: mismatch detected for 'RuntimeLibrary': value 'MTd_StaticDebug' doesn't match value 'MDd_DynamicDebug' in cuda-align.obj
1>ds5-device.obj : error LNK2038: mismatch detected for 'RuntimeLibrary': value 'MTd_StaticDebug' doesn't match value 'MDd_DynamicDebug' in cuda-align.obj
1>ds5-color.obj : error LNK2038: mismatch detected for 'RuntimeLibrary': value 'MTd_StaticDebug' doesn't match value 'MDd_DynamicDebug' in cuda-align.obj
1>ds5-active.obj : error LNK2038: mismatch detected for 'RuntimeLibrary': value 'MTd_StaticDebug' doesn't match value 'MDd_DynamicDebug' in cuda-align.obj
1>ds5-factory.obj : error LNK2038: mismatch detected for 'RuntimeLibrary': value 'MTd_StaticDebug' doesn't match value 'MDd_DynamicDebug' in cuda-align.obj
1>ds5-fw-update-device.obj : error LNK2038: mismatch detected for 'RuntimeLibrary': value 'MTd_StaticDebug' doesn't match value 'MDd_DynamicDebug' in cuda-align.obj
1>rs_advanced_mode.obj : error LNK2038: mismatch detected for 'RuntimeLibrary': value 'MTd_StaticDebug' doesn't match value 'MDd_DynamicDebug' in cuda-align.obj

dorodnic, 2019-09-04:
You can find the relevant flags in visual studio project settings (/mt vs /md) and make sure everything is dynamic debug (seem to be better for your case)
In general this hints at some kind of compilation mismatch (static vs dynamic lib, etc) 

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

nawara72, 2019-09-04:
I simply use the cmakelists provided by librealsense, turn cuda on and generate a sln file from cmake and then compile in visual studio. So I am not really changing any configurations. So not sure why it is not working

RealSenseCustomerSupport, 2019-10-02:
----------------------------------------------

Hi @nawara72,

Do you resolve the build issue after you change the runtime library option from /MT to /MD?

nawara72, 2019-10-02:
Yes I did thank you. Build_with_static_crt in the cmake was turned on, i just had to untick it. Thanks

closed 2019-10-03 by RealSenseCustomerSupport