realsenseai/librealsense #4794: The requested pixel format 'W10 ' is not natively supported by the Linux kernel and likely requires a patch!

hongluck, 2019-09-03:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |    Linux (Ubuntu 16.04)   | 
| Kernel Version (Linux Only)    |   4.15.0-60                                         | 
| Platform                                 | PC|
| SDK Version                            |                        | 
| Language                            |  {C/C#/}                          | 
| Segment			|  {Robot }                          | 

### Issue Description

Hello I hava a D415 Depth Camera and it won't connect to my laptop.
The main error for this is mentioned below.
I can access the realsense viewer without any problems but when ever I turn my node on by using
roslaunch realsense2_camera rs_camera.launch
it gives this error. Please help me thank you

My kernel version is 4.15.0-60-generic

An exception has been thrown: The requested pixel format 'W10 ' is not natively supported by the Linux kernel and likely requires a patch!
Alternatively please upgrade to kernel 4.12 or later. Last Error: Invalid argument
terminate called after throwing an instance of 'rs2::backend_error'
what(): The requested pixel format 'W10 ' is not natively supported by the Linux kernel and likely requires a patch!
Alternatively please upgrade to kernel 4.12 or later. Last Error: Invalid argument
[camera/realsense2_camera_manager-1] process has died [pid 15139, exit code -6, cmd /opt/ros/kinetic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/hong/.ros/log/68c59866-ce8c-11e9-8e37-a86daaec59a0/camera-realsense2_camera_manager-1.log].
log file: /home/hong/.ros/log/68c59866-ce8c-11e9-8e37-a86daaec59a0/camera-realsense2_camera_manager-1*.log

ev-mp, 2019-09-04:
@hongluck , for some reason the request from launch included "W10" infrared format which is used exclusively for calibration, was it on purpose? Are you running a modified launch file?
If yes - please elaborate the use-case and the changes as the Calibration mode has constrains.
If not - can you attach the output if "rs-enumerate-devices -o" ?

hongluck, 2019-09-04:
No, I did not modify it. I was doing the same way as it is explained in the wiki.ros.org following the orders.
The link is below.
https://github.com/IntelRealSense/realsense-ros#installation-instructions

This is the output you asked for.
Thank you for your kind help.

    Infrared 1	  1920x1080	@ 25Hz	   Y16
    Infrared 2	  1920x1080	@ 25Hz	   Y16
    Infrared 2	  1920x1080	@ 25Hz	   Y8
    Infrared 1	  1920x1080	@ 25Hz	   Y8
    Infrared 2	  1920x1080	@ 15Hz	   Y16
    Infrared 1	  1920x1080	@ 15Hz	   Y16
    Infrared 1	  1920x1080	@ 15Hz	   Y8
    Infrared 2	  1920x1080	@ 15Hz	   Y8
    Infrared 1	  1280x720	@ 30Hz	   Y10BPACK
    Infrared	  1280x720	@ 30Hz	   UYVY
    Infrared 1	  1280x720	@ 30Hz	   RAW10
    Infrared 1	  1280x720	@ 30Hz	   Y8
    Infrared 2	  1280x720	@ 30Hz	   Y8
    Infrared	  1280x720	@ 30Hz	   BGRA8
    Infrared	  1280x720	@ 30Hz	   RGBA8
    Infrared	  1280x720	@ 30Hz	   BGR8
    Infrared	  1280x720	@ 30Hz	   RGB8
    Infrared 1	  1280x720	@ 15Hz	   Y10BPACK
    Infrared	  1280x720	@ 15Hz	   UYVY
    Infrared 1	  1280x720	@ 15Hz	   RAW10
    Infrared 1	  1280x720	@ 15Hz	   Y8
    Infrared 2	  1280x720	@ 15Hz	   Y8
    Infrared	  1280x720	@ 15Hz	   BGRA8
    Infrared	  1280x720	@ 15Hz	   RGBA8
    Infrared	  1280x720	@ 15Hz	   BGR8
    Infrared	  1280x720	@ 15Hz	   RGB8
    Infrared 1	  1280x720	@ 6Hz	   Y10BPACK
    Infrared	  1280x720	@ 6Hz	   UYVY
    Infrared 1	  1280x720	@ 6Hz	   RAW10
    Infrared 2	  1280x720	@ 6Hz	   Y8
    Infrared 1	  1280x720	@ 6Hz	   Y8
    Infrared	  1280x720	@ 6Hz	   BGRA8
    Infrared	  1280x720	@ 6Hz	   RGBA8
    Infrared	  1280x720	@ 6Hz	   BGR8
    Infrared	  1280x720	@ 6Hz	   RGB8
    Infrared 2	  960x540	@ 25Hz	   Y16
    Infrared 1	  960x540	@ 25Hz	   Y16
    Infrared 2	  960x540	@ 15Hz	   Y16
    Infrared 1	  960x540	@ 15Hz	   Y16
    Infrared 1	  848x480	@ 90Hz	   Y10BPACK
    Infrared	  848x480	@ 90Hz	   UYVY
    Infrared 1	  848x480	@ 90Hz	   RAW10
    Infrared 1	  848x480	@ 90Hz	   Y8
    Infrared 2	  848x480	@ 90Hz	   Y8
    Infrared	  848x480	@ 90Hz	   BGRA8
    Infrared	  848x480	@ 90Hz	   RGBA8
    Infrared	  848x480	@ 90Hz	   BGR8
    Infrared	  848x480	@ 90Hz	   RGB8
    Infrared 1	  848x480	@ 60Hz	   Y10BPACK
    Infrared	  848x480	@ 60Hz	   UYVY
    Infrared 1	  848x480	@ 60Hz	   RAW10
    Infrared 1	  848x480	@ 60Hz	   Y8
    Infrared 2	  848x480	@ 60Hz	   Y8
    Infrared	  848x480	@ 60Hz	   BGRA8
    Infrared	  848x480	@ 60Hz	   RGBA8
    Infrared	  848x480	@ 60Hz	   BGR8
    Infrared	  848x480	@ 60Hz	   RGB8
    Infrared 1	  848x480	@ 30Hz	   Y10BPACK
    Infrared	  848x480	@ 30Hz	   UYVY
    Infrared 1	  848x480	@ 30Hz	   RAW10
    Infrared 2	  848x480	@ 30Hz	   Y8
    Infrared 1	  848x480	@ 30Hz	   Y8
    Infrared	  848x480	@ 30Hz	   BGRA8
[... 410 lines cut ...]
    Color	  320x240	@ 30Hz	   YUYV
    Color	  320x240	@ 6Hz	   RGB8
    Color	  320x240	@ 6Hz	   Y16
    Color	  320x240	@ 6Hz	   BGRA8
    Color	  320x240	@ 6Hz	   RGBA8
    Color	  320x240	@ 6Hz	   BGR8
    Color	  320x240	@ 6Hz	   YUYV
    Color	  320x180	@ 60Hz	   RGB8
    Color	  320x180	@ 60Hz	   Y16
    Color	  320x180	@ 60Hz	   BGRA8
    Color	  320x180	@ 60Hz	   RGBA8
    Color	  320x180	@ 60Hz	   BGR8
    Color	  320x180	@ 60Hz	   YUYV
    Color	  320x180	@ 30Hz	   RGB8
    Color	  320x180	@ 30Hz	   Y16
    Color	  320x180	@ 30Hz	   BGRA8
    Color	  320x180	@ 30Hz	   RGBA8
    Color	  320x180	@ 30Hz	   BGR8
    Color	  320x180	@ 30Hz	   YUYV
    Color	  320x180	@ 6Hz	   RGB8
    Color	  320x180	@ 6Hz	   Y16
    Color	  320x180	@ 6Hz	   BGRA8
    Color	  320x180	@ 6Hz	   RGBA8
    Color	  320x180	@ 6Hz	   BGR8
    Color	  320x180	@ 6Hz	   YUYV

hongluck, 2019-09-04:
I found out that while I was running realsense viewer, it was working fine but there were error messages as well.
04/09 10:13:04,591 WARNING [140333542790912] (backend-v4l2.cpp:1148) Pixel format 20303157-0000-0010-8000-00aa003 likely requires patch for fourcc code W10 !
 04/09 10:13:04,599 WARNING [140333542790912] (backend-v4l2.cpp:1148) Pixel format 36315752-1a66-a242-9065-d01814a likely requires patch for fourcc code RW16!
 04/09 10:13:27,952 WARNING [140333417490176] (ds5-timestamp.cpp:64) UVC metadata payloads not available. Please refer to the installation chapter for details.
 04/09 10:13:33,829 WARNING [140333400704768] (ds5-timestamp.cpp:64) UVC metadata payloads not available. Please refer to the installation chapter for details.

hongluck, 2019-09-05:
These are some error messages I have found whenever I turn my PC on.
Also I think it is related to a graphic driver problem as well. I will leave the link.

1. Error messages including librealsense
[https://imgur.com/a/EI55m6v](url)
2. gtx1650 graphic driver issues 
[https://devtalk.nvidia.com/default/topic/1061656/linux/gtx1650-notebook-not-working-on-ubuntu16-04-black-screen-amp-login-loop-/3](url)

MarqRazz, 2019-09-05:
I get the same error
`An exception has been thrown: The requested pixel format 'W10 ' is not natively supported by the Linux kernel and likely requires a patch!
Alternatively please upgrade to kernel 4.12 or later. Last Error: Invalid argument`

My kernel version is: 4.15.0-60-generic

Here is my output from `rs-enumerate-devices -o`

`Device info: 
    Name                          : 	Intel RealSense D415
    Serial Number                 : 	746112061346
    Firmware Version              : 	05.11.11.100
    Recommended Firmware Version  : 	05.11.11.100
    Physical Port                 : 	/sys/devices/pci0000:00/0000:00:14.0/usb2/2-8/2-8:1.0/video4linux/video0
    Debug Op Code                 : 	15
    Advanced Mode                 : 	YES
    Product Id                    : 	0AD3
    Camera Locked                 : 	YES
    Usb Type Descriptor           : 	3.2
    Product Line                  : 	D400
    Asic Serial Number            : 	736313020554

Options for Stereo Module
 Supported options:                min        max       step  default  
    Exposure                       : 1    ... 165000      1     33000     
    Gain                           : 16   ... 248         1     16        
    Enable Auto Exposure           : 0    ... 1           1     1         
    Enable Auto White Balance      : 0    ... 1           1     0         
    Visual Preset                  : 0    ... 6           1     0         
    Laser Power                    : 0    ... 360         30    150       
    Emitter Enabled                : 0    ... 1           1     1         
    Frames Queue Size              : 0    ... 32          1     16        
    Error Polling Enabled          : 0    ... 1           1     0         
    Output Trigger Enabled         : 0    ... 1           1     0         
    Depth Units                    : 0.0001... 0.01        1e-06 0.001     
    Stereo Baseline                : 55.062... 55.062      0     55.062    
    Inter Cam Sync Mode            : 0    ... 2           0     1         
    Global Time Enabled            : 0    ... 1           1     1         

Options for RGB Camera
 Supported options:                min        max       step  default  
    Backlight Compensation         : 0    ... 1           1     0         
    Brightness                     : -64  ... 64          1     0         
    Contrast                       : 0    ... 100         1     50        
    Exposure                       : 1    ... 10000       1     166       
    Gain                           : 0    ... 128         1     64        
    Gamma                          : 100  ... 500         1     300       
    Hue                            : -180 ... 180         1     0         
    Saturation                     : 0    ... 100         1     64        
    Sharpness                      : 0    ... 100         1     50        
    White Balance                  : 2800 ... 6500        10    4600      
    Enable Auto Exposure           : 0    ... 1           1     1         
    Enable Auto White Balance      : 0    ... 1           1     1         
    Frames Queue Size              : 0    ... 32          1     16        
    Power Line Frequency           : 0    ... 2           1     3         
    Auto Exposure Priority         : 0    ... 1           1     0         
    Global Time Enabled            : 0    ... 1           1     1         

Stream Profiles supported by Stereo Module
 Supported modes:
    stream       resolution      fps       format   
    Infrared 2	  1920x1080	@ 25Hz	   Y16
    Infrared 1	  1920x1080	@ 25Hz	   Y16
    Infrared 1	  1920x1080	@ 25Hz	   Y8
    Infrared 2	  1920x1080	@ 25Hz	   Y8
    Infrared 2	  1920x1080	@ 15Hz	   Y16
    Infrared 1	  1920x1080	@ 15Hz	   Y16
    Infrared 1	  1920x1080	@ 15Hz	   Y8
    Infrared 2	  1920x1080	@ 15Hz	   Y8
    Infrared 1	  1280x720	@ 30Hz	   Y10BPACK
[... 465 lines cut ...]
    Color	  320x240	@ 6Hz	   RGB8
    Color	  320x240	@ 6Hz	   Y16
    Color	  320x240	@ 6Hz	   BGRA8
    Color	  320x240	@ 6Hz	   RGBA8
    Color	  320x240	@ 6Hz	   BGR8
    Color	  320x240	@ 6Hz	   YUYV
    Color	  320x180	@ 60Hz	   RGB8
    Color	  320x180	@ 60Hz	   Y16
    Color	  320x180	@ 60Hz	   BGRA8
    Color	  320x180	@ 60Hz	   RGBA8
    Color	  320x180	@ 60Hz	   BGR8
    Color	  320x180	@ 60Hz	   YUYV
    Color	  320x180	@ 30Hz	   RGB8
    Color	  320x180	@ 30Hz	   Y16
    Color	  320x180	@ 30Hz	   BGRA8
    Color	  320x180	@ 30Hz	   RGBA8
    Color	  320x180	@ 30Hz	   BGR8
    Color	  320x180	@ 30Hz	   YUYV
    Color	  320x180	@ 6Hz	   RGB8
    Color	  320x180	@ 6Hz	   Y16
    Color	  320x180	@ 6Hz	   BGRA8
    Color	  320x180	@ 6Hz	   RGBA8
    Color	  320x180	@ 6Hz	   BGR8
    Color	  320x180	@ 6Hz	   YUYV
`

hongluck, 2019-09-05:
> I get the same error
> `An exception has been thrown: The requested pixel format 'W10 ' is not natively supported by the Linux kernel and likely requires a patch! Alternatively please upgrade to kernel 4.12 or later. Last Error: Invalid argument`
> 
> My kernel version is: 4.15.0-60-generic
> 
> I'm running Ubuntu 18.04 and have recently purged and reinstalled librealsense trying to fix the error.
> 
> Here is my output from `rs-enumerate-devices -o`
> 
> `Device info:
> Name : Intel RealSense D415
> Serial Number : 746112061346
> Firmware Version : 05.11.11.100
> Recommended Firmware Version : 05.11.11.100
> Physical Port : /sys/devices/pci0000:00/0000:00:14.0/usb2/2-8/2-8:1.0/video4linux/video0
> Debug Op Code : 15
> Advanced Mode : YES
> Product Id : 0AD3
> Camera Locked : YES
> Usb Type Descriptor : 3.2
> Product Line : D400
> Asic Serial Number : 736313020554
> 
> Options for Stereo Module
> Supported options: min max step default
> Exposure : 1 ... 165000 1 33000
> Gain : 16 ... 248 1 16
> Enable Auto Exposure : 0 ... 1 1 1
> Enable Auto White Balance : 0 ... 1 1 0
> Visual Preset : 0 ... 6 1 0
> Laser Power : 0 ... 360 30 150
> Emitter Enabled : 0 ... 1 1 1
> Frames Queue Size : 0 ... 32 1 16
> Error Polling Enabled : 0 ... 1 1 0
> Output Trigger Enabled : 0 ... 1 1 0
> Depth Units : 0.0001... 0.01 1e-06 0.001
> Stereo Baseline : 55.062... 55.062 0 55.062
> Inter Cam Sync Mode : 0 ... 2 0 1
> Global Time Enabled : 0 ... 1 1 1
> 
> Options for RGB Camera
> Supported options: min max step default
> Backlight Compensation : 0 ... 1 1 0
> Brightness : -64 ... 64 1 0
> Contrast : 0 ... 100 1 50
> Exposure : 1 ... 10000 1 166
> Gain : 0 ... 128 1 64
> Gamma : 100 ... 500 1 300
> Hue : -180 ... 180 1 0
> Saturation : 0 ... 100 1 64
> Sharpness : 0 ... 100 1 50
> White Balance : 2800 ... 6500 10 4600
> Enable Auto Exposure : 0 ... 1 1 1
> Enable Auto White Balance : 0 ... 1 1 1
> Frames Queue Size : 0 ... 32 1 16
> Power Line Frequency : 0 ... 2 1 3
> Auto Exposure Priority : 0 ... 1 1 0
> Global Time Enabled : 0 ... 1 1 1
> 
> Stream Profiles supported by Stereo Module
> Supported modes:
> stream resolution fps format
> Infrared 2 1920x1080 @ 25Hz Y16
> Infrared 1 1920x1080 @ 25Hz Y16
> Infrared 1 1920x1080 @ 25Hz Y8
> Infrared 2 1920x1080 @ 25Hz Y8
> Infrared 2 1920x1080 @ 15Hz Y16
> Infrared 1 1920x1080 @ 15Hz Y16
> Infrared 1 1920x1080 @ 15Hz Y8
> Infrared 2 1920x1080 @ 15Hz Y8
[... 468 lines cut ...]
> Color 320x240 @ 6Hz BGRA8
> Color 320x240 @ 6Hz RGBA8
> Color 320x240 @ 6Hz BGR8
> Color 320x240 @ 6Hz YUYV
> Color 320x180 @ 60Hz RGB8
> Color 320x180 @ 60Hz Y16
> Color 320x180 @ 60Hz BGRA8
> Color 320x180 @ 60Hz RGBA8
> Color 320x180 @ 60Hz BGR8
> Color 320x180 @ 60Hz YUYV
> Color 320x180 @ 30Hz RGB8
> Color 320x180 @ 30Hz Y16
> Color 320x180 @ 30Hz BGRA8
> Color 320x180 @ 30Hz RGBA8
> Color 320x180 @ 30Hz BGR8
> Color 320x180 @ 30Hz YUYV
> Color 320x180 @ 6Hz RGB8
> Color 320x180 @ 6Hz Y16
> Color 320x180 @ 6Hz BGRA8
> Color 320x180 @ 6Hz RGBA8
> Color 320x180 @ 6Hz BGR8
> Color 320x180 @ 6Hz YUYV
> `

Are you using the gtx1650 graphic card as well?

MarqRazz, 2019-09-05:
No just a standard i7 with a internal 630 graphic's card

doronhi, 2019-09-16:
The problem occurs by realsense2_camera choosing the input format from the camera without specifying the input format. In the latest frimware update a new format was introduced, W10, which is not supported by the default kernel.
To fix the problem with kernel 4.15.0-60 run the following script patch-realsense-ubuntu-lts.sh. If you installed librealsense from debian packages you will need to follow the manual installation guide [here](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md) up until the section of "Build and apply patched kernel modules" and then run the script. Pay attention, If you have kernel 4.15.0-60 you need to run `patch-realsense-ubuntu-lts.sh` and *not* `patch-ubuntu-kernel-4.16.sh`.

In addition, I will submit a fix to realsense2_camera to avoid this error.

doronhi, 2019-09-16:
fixed in [realsense2_camera PR #924](https://github.com/IntelRealSense/realsense-ros/pull/924)

fgrzadkowski, 2019-09-16:
@doronhi I'm facing a similar issue, with:
* SDK version 2.25.0-0~realsense0.1362
* Realsense ROS version 2.2.8
* Ubuntu 16.04
* Kernel: 4.4.0-131-generic
* Nvidia driver version: 390.48
* DKMS version: 1.3.4-0ubuntu1

I'm using nvidia gtx 1080. I installed everything from apt packages, and I see the following error:

```
A 2019-09-13T07:56:02.296917635Z  13/09 07:56:02,296 WARNING [140034679183104] (backend-v4l2.cpp:1149) Pixel format 20303157-0000-0010-8000-00aa003 likely requires patch for fourcc code W10 ! 
A 2019-09-13T07:56:02.298548176Z  13/09 07:56:02,298 ERROR [140034679183104] (types.h:243) The requested pixel format 'W10 ' is not natively supported by the Linux kernel and likely requires a patch! 
A 2019-09-13T07:56:02.298550162Z Alternatively please upgrade to kernel 4.12 or later. Last Error: Invalid argument 
A 2019-09-13T07:56:02.299257939Z  [31m[ERROR] [1568361362.299137722]: An exception has been thrown: The requested pixel format 'W10 ' is not natively supported by the Linux kernel and likely requires a patch! 
A 2019-09-13T07:56:02.299259868Z Alternatively please upgrade to kernel 4.12 or later. Last Error: Invalid argument [0m 
A 2019-09-13T07:56:02.299416264Z terminate called after throwing an instance of 'rs2::backend_error' 
A 2019-09-13T07:56:02.299436137Z   what():  The requested pixel format 'W10 ' is not natively supported by the Linux kernel and likely requires a patch! 
A 2019-09-13T07:56:02.299436648Z Alternatively please upgrade to kernel 4.12 or later. Last Error: Invalid argument 
A 2019-09-13T07:56:02.486309146Z  [31m[camera_cart_left/realsense2_camera_manager-1] process has died [pid 104, exit code -6, cmd /opt/ros/kinetic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/root/.ros/log/bbbf4e56-d5fb-11e9-b5e1-a0369f04326f/camera_cart_left-realsense2_camera_manager-1.log]. 
A 2019-09-13T07:56:02.486311496Z log file: /root/.ros/log/bbbf4e56-d5fb-11e9-b5e1-a0369f04326f/camera_cart_left-realsense2_camera_manager-1*.log [0m 
A 2019-09-13T07:56:02.975468492Z  [1mprocess[box_placing_depth_baseline_srv-1]: started with pid [438] [0m 
A 2019-09-13T07:56:02.976452294Z  [1mprocess[camera_calibration_downloader_eddie_333_1560272407429094588-1]: started with pid [444] [0m 
A 2019-09-13T07:56:02.976766641Z  [1mprocess[robot_state_publisher-1]: started with pid [447] [0m 
A 2019-09-13T07:56:03.077657230Z shutting down processing monitor... 
A 2019-09-13T07:56:03.077775899Z ... shutting down processing monitor complete 
A 2019-09-13T07:56:03.077818606Z  [1mdone [0m 
A 2019-09-13T07:56:03.200225446Z  [34mSuccessful exit from subprocess: ['roslaunch', 'nomagic_launch', 'camera_calibration_downloader.launch', 'camera_id:=box_placing', 'allow_no_internet:=False', 'camera_serial_no:=810512061758'] [0m 
```

What is interesting I don't see this error on another system with the same spec (or at least I failed to find differences).

Any advice?

fgrzadkowski, 2019-09-16:
<details>
<summary>
Output from `rs-enumerate-devices -o` (click on "arrow" on the left to unfold as it's pretty long):
</summary>
$ rs-enumerate-devices -o
 Device info: 
    Name                          : 	Intel RealSense D415
    Serial Number                 : 	746112061823
    Firmware Version              : 	05.11.10.00
    Recommended Firmware Version  : 	05.09.14.00
    Physical Port                 : 	/sys/devices/pci0000:b2/0000:b2:00.0/0000:b3:00.0/0000:b4:04.0/0000:b8:00.0/usb10/10-1/10-1:1.0/video4linux/video6
    Debug Op Code                 : 	15
    Advanced Mode                 : 	YES
    Product Id                    : 	0AD3
    Usb Type Descriptor           : 	3.2

Options for Stereo Module
 Supported options:                                    min        max       step  default  
    Exposure                                           : 1    ... 165000      1     33000     
    Gain                                               : 16   ... 248         1     16        
    Enable Auto Exposure                               : 0    ... 1           1     1         
    Enable Auto White Balance                          : 0    ... 1           1     0         
    Visual Preset                                      : 0    ... 6           1     0         
    Laser Power                                        : 0    ... 360         30    150       
    Emitter Enabled                                    : 0    ... 1           1     1         
    Frames Queue Size                                  : 0    ... 32          1     16        
    Error Polling Enabled                              : 0    ... 1           1     0         
    Output Trigger Enabled                             : 0    ... 1           1     0         
    Depth Units                                        : 0.0001... 0.01        1e-06 0.001     
    Stereo Baseline                                    : 54.9737... 54.9737     0     54.9737   
    Inter Cam Sync Mode                                : 0    ... 2           0     1         

Options for RGB Camera
 Supported options:                                    min        max       step  default  
    Backlight Compensation                             : 0    ... 1           1     0         
    Brightness                                         : -64  ... 64          1     0         
    Contrast                                           : 0    ... 100         1     50        
    Exposure                                           : 1    ... 10000       1     166       
    Gain                                               : 0    ... 128         1     64        
    Gamma                                              : 100  ... 500         1     300       
    Hue                                                : -180 ... 180         1     0         
    Saturation                                         : 0    ... 100         1     64        
    Sharpness                                          : 0    ... 100         1     50        
    White Balance                                      : 2800 ... 6500        10    4600      
    Enable Auto Exposure                               : 0    ... 1           1     1         
    Enable Auto White Balance                          : 0    ... 1           1     1         
    Frames Queue Size                                  : 0    ... 32          1     16        
    Power Line Frequency                               : 0    ... 3           1     3         
    Auto Exposure Priority                             : 0    ... 1           1     0         

Stream Profiles supported by Stereo Module
 Supported modes:                                      stream     resolution fps   format   
    Infrared 1	  1920x1080	@ 25Hz	Y16
    Infrared 2	  1920x1080	@ 25Hz	Y16
    Infrared 2	  1920x1080	@ 25Hz	Y8
    Infrared 1	  1920x1080	@ 25Hz	Y8
    Infrared 2	  1920x1080	@ 15Hz	Y16
    Infrared 1	  1920x1080	@ 15Hz	Y16
    Infrared 1	  1920x1080	@ 15Hz	Y8
    Infrared 2	  1920x1080	@ 15Hz	Y8
    Infrared	  1280x720	@ 30Hz	UYVY
    Infrared 2	  1280x720	@ 30Hz	Y8
    Infrared 1	  1280x720	@ 30Hz	Y8
    Infrared	  1280x720	@ 30Hz	BGRA8
    Infrared	  1280x720	@ 30Hz	RGBA8
    Infrared	  1280x720	@ 30Hz	BGR8
    Infrared	  1280x720	@ 30Hz	RGB8
    Infrared	  1280x720	@ 15Hz	UYVY
    Infrared 2	  1280x720	@ 15Hz	Y8
    Infrared 1	  1280x720	@ 15Hz	Y8
[... 1374 lines cut ...]
    Color	  320x240	@ 6Hz	RGB8
    Color	  320x240	@ 6Hz	Y16
    Color	  320x240	@ 6Hz	BGRA8
    Color	  320x240	@ 6Hz	RGBA8
    Color	  320x240	@ 6Hz	BGR8
    Color	  320x240	@ 6Hz	YUYV
    Color	  320x180	@ 60Hz	RGB8
    Color	  320x180	@ 60Hz	Y16
    Color	  320x180	@ 60Hz	BGRA8
    Color	  320x180	@ 60Hz	RGBA8
    Color	  320x180	@ 60Hz	BGR8
    Color	  320x180	@ 60Hz	YUYV
    Color	  320x180	@ 30Hz	RGB8
    Color	  320x180	@ 30Hz	Y16
    Color	  320x180	@ 30Hz	BGRA8
    Color	  320x180	@ 30Hz	RGBA8
    Color	  320x180	@ 30Hz	BGR8
    Color	  320x180	@ 30Hz	YUYV
    Color	  320x180	@ 6Hz	RGB8
    Color	  320x180	@ 6Hz	Y16
    Color	  320x180	@ 6Hz	BGRA8
    Color	  320x180	@ 6Hz	RGBA8
    Color	  320x180	@ 6Hz	BGR8
    Color	  320x180	@ 6Hz	YUYV
</details>

doronhi, 2019-09-16:
Did you try [realsense2_camera's PR#924](https://github.com/IntelRealSense/realsense-ros/pull/924)?

hongluck, 2019-09-16:
@doronhi  thank you it is now working fine.
You are the boss. Have a wonderful day

closed 2019-09-16 by hongluck

fgrzadkowski, 2019-09-16:
@doronhi It turned out to be a problem with not properly installed DKMS for the driver. Thanks!

newtaste, 2019-09-17:
I am also struggling with the same problem in the same situation.
Can you tell me a little bit more about how to fix this?

ev-mp, 2019-09-18:
@newtaste , the issue is specific to ROS users and if you're affected then it means that you're using unpatched kernel and need to apply the kernel patches. the SDK offers two ways to handle this - DKMS or installation script:
1) DKMS can be applied to Ubuntu LTS kernels. 
>sudo apt install librealsense2-dkms

Note that there is a known issue with the 4.15 branch (#4669) that affects versions 4.15.0-58+. In that case it is safer to use the second method.

2) [Installation script](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md)
>./scripts/patch-realsense-ubuntu-lts.sh