realsenseai/librealsense #4795: Error realsense viewer HRESULT: 0x8007001F

visus100, 2019-09-03:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |D435i | 
| Firmware Version                   | FW_5_11_11_100 | 
| Operating System & Version |   Win 10 | 
| Platform                                 | PC|
| SDK Version                            |  SDK-WIN10-2.23.0.909                        | 
| Language                            |  C++                  | 

### Issue Description

After trying to change preset in realsense viewer i got error:

[image]

Also i get this error when i'm trying to change in Stereo module > controls > gain (value)
Other settings works fine, and cameras streaming well.

Backend in rs2_set_option(options:0543E018, option:Gain, value:35):
hr returned: HResult 0x8007001f:

I tried reboot PC, change USB Port, also i own 2 of same scanners and on both same issues.
What more it works well 1 week ago, but now i am getting this strange errors.

Whats more i have same issues when i'm compiling my own code when changing profile preset by visual studio 2019

### Any ideas how to solve problem?

ev-mp, 2019-09-26:
@visus100 hello, sorry for the late response.
The version you're using 2.19 is rather outdated. There were numerous enhancements and bug fixes introduced, including Window OS.
Please upgrade the SDK to one of the recent versions (2.28+) and check whether this error is still reproducible.

RealSenseCustomerSupport, 2019-10-12:
----------------------------------------------

@visus100 Did you get chance to try our latest SDK? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2019-10-16:
----------------------------------------------

@visus100 Could you please update if you still have issue on this? Thanks!

RealSenseCustomerSupport, 2019-10-25:
----------------------------------------------

@visus100 Sorry that we didn't get response from you for weeks. Will close the ticket at this point. Please feel free to create new one if you still have questions/issues. Thanks!

closed 2019-10-25 by RealSenseCustomerSupport