realsenseai/librealsense #4796: Android Pointcloud - mapTo/calculate

erikgallegos, 2019-09-04:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 05.11.01.100 | 
| Operating System & Version |   Android 8.0  | 
| Platform                                 | Galaxy S8 |
| SDK Version                            |  2.25                         | 
| Language                            |  kotlin                       | 
| Segment			| Smartphone    | 

### Issue Description

I need to use `Pointcloud`'s `mapTo` and `calculate` functions but they are not currently available for Android. I tried adding the functionality myself, as seen below, but my app crashes whenever either function is called. I believe the error might be caused by incorrectly casting something but I'm not entirely sure. Are there plans to add this functionality anytime soon? If not, I'd be happy to submit a pull request, given some help to get it working properly.

```
extern "C" JNIEXPORT void JNICALL
Java_com_intel_realsense_librealsense_Pointcloud_nMapTo(JNIEnv *env, jclass type, jlong handle, jlong mapped) {
    rs2_error *e = NULL;
    auto pointCloud = reinterpret_cast<rs2::pointcloud *>(handle);
    rs2::frame mappedFrame = reinterpret_cast<rs2_frame *>(mapped);
    pointCloud->map_to(mappedFrame);
    handle_error(env, e);
}

extern "C" JNIEXPORT void JNICALL
Java_com_intel_realsense_librealsense_Pointcloud_nCalculate(JNIEnv *env, jclass type, jlong handle, jlong depthFrame, jfloatArray data) {
    rs2_error *e = NULL;
    jsize length = env->GetArrayLength(data);
    auto pointCloud = reinterpret_cast<rs2::pointcloud *>(handle);
    rs2::frame frame = reinterpret_cast<rs2_frame *>(depthFrame);
    auto points = pointCloud->calculate(frame);
    env->SetFloatArrayRegion(data, 0, length, reinterpret_cast<const jfloat *>(rs2_get_frame_data(reinterpret_cast<const rs2_frame *>(&points), &e)));
    handle_error(env, e);
}
````

matkatz, 2019-09-04:
please see #4601

erikgallegos, 2019-09-04:
@matkatz I see how the pointcloud vertices can be obtained but I don't feel that my question has been answered. Ideally I would like to replicate this workflow from matlab:

```
pointcloud_var = realsense.pointcloud();
align_var = realsense.align(2);
spatial = realsense.spatial_filter(0.5,20.0,2.0,5);
temporal = realsense.temporal_filter(0.4,20.0,2);

frameSet = pipe.wait_for_frames();
aligned = align_var.process(frameSet);
depth = aligned.get_depth_frame();
depth = spatial.process(depth);
depth = temporal.process(depth);
color = aligned.get_color_frame();

pointcloud_var.map_to(color);
points = pointcloud_var.calculate(depth); 
vertices = points.get_vertices();
```

With the end goal being an array of pointcloud vertices that are aligned/mapped with the color frame.

matkatz, 2019-09-09:
`calculate` is deprecated function and it currently simply calls to `process` in C++ wrapper.

`map_to` can be added but in case you work with pipeline and frameset you don't need it, simply provide the texture stream to the point cloud filter:
Pointcloud mPointcloud= new Pointcloud(StreamType.COLOR);

Please notice that align filter is not required if pointcloud filter is used, the alignment can done be using the texture coordinates.
If you need it for other reasons, simply add it to the pipe below.

You can concatenate the processing blocks as follow using apply filter API:
```
try(FrameReleaser fr = new FrameReleaser()){
    FrameSet frames = mPipeline.waitForFrames().releaseWith(fr);
    FrameSet processedSet = frames.applyFilter(mDecimationFilter).releaseWith(fr).
                                             applyFilter(mTemporalFilter).releaseWith(fr).
                                             applyFilter(mSpatialFilter).releaseWith(fr).
                                             applyFilter(mPointcloud).releaseWith(fr);
    Frame processed = processedSet.first(StreamType.DEPTH, StreamFormat.XYZ32F).releaseWith(fr);
    Frame texture = processedSet.first(StreamType.COLOR, StreamFormat.RGB8).releaseWith(fr);
    Points points = processed.as(Extension.POINTS);
    float[] ver = points.getVertices();
    float[] texMap = points.getTextureCoordinates();
}
```
Check `createTexture` (line 90) in [GLPointsFrame](https://github.com/matkatz/librealsense/blob/android-pointcloud-support/wrappers/android/librealsense/src/main/java/com/intel/realsense/librealsense/GLPointsFrame.java) for example how to align the points with the texture.
 
Please see [processing example](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/android/examples/processing/src/main/java/com/intel/realsense/processing/MainActivity.java) for more details.

matkatz, 2019-09-11:
No, a frame with stream format XYZ32F is added to the frameset that you apply the pointcloud filter on.
This is the expected behavior:
Pointcloud mPointcloudFilter = new Pointcloud(StreamType.COLOR);
FrameSet originalSet = mPipeline.waitForFrames();
Frame p1 = originalSet .first(StreamType.DEPTH, StreamFormat.XYZ32F); 
// p1 one is null since pointcloud filter wasn't applied.
FrameSet frameSetWithPointcloud = frames.applyFilter(mPointcloudFilter);
Frame p2 = frameSetWithPointcloud.first(StreamType.DEPTH, StreamFormat.XYZ32F);
// p2 one is a valid points frame.
Points pc = p2.as(Extension.POINTS);
int c = pc.getCount();

There were some API changes lately, please use release 2.28.0 or the development branch.