realsenseai/librealsense #4797: find the point has min depth value in color image

tonyjonyjack, 2019-09-04:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435  | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Win (10) | 
| Platform                                 | PC  |
| SDK Version                            |   legacy / 2.0                           | 
| Language                            |  C++/opencv                         | 

### Issue Description

### Issue Description
<Describe your issue / question / feature request / etc..>
I try to find the point in color image  closest to the camera.
so I transform every point in colocr image to depth map,then I get the distance between every point  and camera.But the result are terrible,the computer run very slow,20-second delay,and the point obviously are not the point closest to the camera.
Thanks in advance.

ev-mp, 2019-09-05:
@tonyjonyjack hello,
To speedup the processing it seem that it will be much more efficient to first align depth to color, or color to depth. This will allow to perform the lookup into 2D frame rather than in 3D space and also take advantage of the already available optimizations, such as SSE
.
Secondly, I would suggest to further improve the algorithm by running the raw depth via decimation filter with a high downsample factor (4+) prior to processing. The reasons are:
- Less data to process in alignment/final search. Running decimation + alignment processing blocks is much faster than running alignment only using unmodified depth.
- Removes "No Data" pixels (hole-filling), which also save processing time.
- But most important - each "decimated" pixel represent an average/mean of 16-64 raw depth pixels so it drastically reduces the intrinsic RMS error and also automatically marks the "center of mass" of the designated area, which facilitates objects recognition.

it may introduce some artifacts, such as it may miss really small objects ( between  [4X4.] and [.8X8] pixels size),

ev-mp, 2019-09-18:
@tonyjonyjack hi, did you manage to find a solution to your use-case ? Or do you need more assistance  - please update.

closed 2019-09-26 by ev-mp