realsenseai/librealsense #4798: Measure metrics of a 3d object

jssa62, 2019-09-04:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 

| Operating System & Version |   {Win (10) / Linux (Ubuntu 16)   | 
| Language                            |  {C/python }                          | 

Hello,
I am trying to measure the metrics (diameter, length and width) of an object (like a pen). 
The goal is to take a 3d picture of the object and create a 3d cloud point and measure the volume of that object.
Do we have any source code or any suggestion for that kind of thing?
Cheers

MartyG-RealSense, 2019-09-04:
For the kind of 3D measuring that you are seeking, the Python-based **Box Dimensioner** sample may provide suitable results, though it is usually used with a minimum of two cameras.

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python/examples/box_dimensioner_multicam

In the Box Dimensioner example, you set up a chessboard object to calibrate the camera, and then place on the board an object that is smaller than the board (so it fits inside the borders of the chessboard).  So a pen should fit fine.

jssa62, 2019-09-16:
@MartyG-RealSense Many thanks for your answer. But this method is only used with a chessboard while in a real-life situation, there is no chessboard. On the other hand, if we want to measure a "T" shaped object, it cannot calculate the width of " I " part.

MartyG-RealSense, 2019-09-17:
I don't doubt that 3D measurement without a chessboard is possible, since it was done even with the original RealSense cameras in 2014 (a Chinese company called JD.com did a warehouse box measuring trial) and it is also a key application of rugged Windows 10 tablets with rear-facing D410 and D415 cameras built into them.  It is likely just a matter of finding software / programs that can do it.

https://support.intelrealsense.com/hc/en-us/community/posts/360033966114-DT-Research-article-on-using-RealSense-in-rugged-tablets-and-other-tablet-suppliers

RealSenseCustomerSupport, 2019-10-08:
----------------------------------------------

Please let us know if it is alright to close this ticket if there's no further question.
Thank you.

closed 2019-10-15 by RealSenseCustomerSupport