realsenseai/librealsense #4799: Some options and controls cannot be set over libuvc on raspberry pi 4

SirDifferential, 2019-09-04:
----------------------------------------------------------------------------------------------------
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.11.11.100 | 
| Operating System & Version |  Raspbian | 
| Kernel Version (Linux Only)    | 4.19.66-v7l+ | 
| Platform                                 | Raspberry Pi 4  |
| SDK Version                            | 2.27  | 
| Language                            | C++ | 
| Segment			| others | 

### Issue Description

Setting some of the more advanced controls through librealsense built with `FORCE_LIBUVC=ON` does not seem to work on raspberry pi 4 raspbian. In all cases the error returned is some variation of:

```
RealSense error calling rs2_set_option(options:0xb26a1b90, option:Emitter Enabled, value:1): set_xu(id=2) failed! Last Error: Device or resource busy
```

The tested requests that don't work are:

* `rs2::sensor::set_option(RS2_OPTION_EMITTER_ENABLED, 1.0f);`
* `rs2::sensor::set_option(RS2_OPTION_LASER_POWER, laserpower.max);`
* `rs400::advanced_mode::set_depth_control(gr);`

Using the non-libuvc version sometimes works (I've noticed you may have to try the request several times), but there are currently no kernel patches for raspbian, and the produced depth image is mostly black probably due to the depth image format not being understood by the drivers. The same requests work on an Ubuntu 18 with the kernel patches applied and produce a valid depth image.

Otherwise, if these requests aren't made, libuvc appears to work on raspbian and produces a valid depth image, although there are occasionally crashes as reported in my other issue.

Here's a simple C++ program to try this: https://github.com/SirDifferential/minimal_rs_advancedmode/blob/master/main.cpp

dorodnic, 2019-11-11:
Should be resolved now (with the latest release)

SirDifferential, 2019-11-11:
Confirmed working on latest master. Fantastic work. :elephant:

closed 2019-11-11 by SirDifferential