realsenseai/librealsense #4800: Fatal unmanaged exception after multiple rs2::pipeline instanciations

blackccpie, 2019-09-04:
-------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 5.11.1 | 
| Operating System & Version |   Win 10  | 
| Platform                                 | PC  |
| SDK Version                            |  2.25.0                          | 
| Language                            |  C++  (Visual Studio 2017)                        | 
| Segment			| others                          | 

### Issue Description
When I repeat multiple consecutive instanciations of rs2::pipeline object in a loop, in my case I always encounter unmanaged fatal exception after 128 iterations. The code I'm using is as simple as that:

```cpp
for(int i=0; i<200; i++)
{
	std::cout << "#" << i << std::endl;
	rs2::pipeline pipe;
}
```

I encountered this problem after implementing internal robustness tests with multiple consecutive camera open/close. This issue is a blocking one for us, because it implies a crash of our application after some time of use...

Could you confirm and help me resolve the problem?

Regards

Albert

SirDifferential, 2019-09-04:
I tried running this with 7a3ffaf412401849894a4d69791b98559f624d47 and don't see any errors on Debian 10. The full program:

```
#include <iostream>
#include <librealsense2/rs.hpp>

int main(int argc, char** argv) {

    for (int i=0; i<200; i++) {
        std::cout << "#" << i << std::endl;
        rs2::pipeline pipe;
    }

    return 0;
}
```

Maybe this is Windows specific?

blackccpie, 2019-09-04:
@SirDifferential : seems like it's Windows specific indeed. I also ran the loop on MacOS and had no problem.

blackccpie, 2019-09-20:
Seems like this issue has been fixed with latest pre-release 2.28.0.

closed 2019-10-04 by blackccpie