realsenseai/librealsense #4801: Turn enable_record_to_file() on/off

bernland, 2019-09-04:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 05.11.10.00 | 
| Operating System & Version |   Win 10  | 
| Platform                                 | PC |
| SDK Version                            |  2.18.0.515                           | 
| Language                            |  python                           | 
| Segment			|  VR/AR                          | 

### Issue Description
Here's what I'm trying to achieve: I'd like to save a single (depth) frame in PLY format and a BAG stream a few seconds before and after that frame. I understand that enable_record_to_file() allows for capturing the BAG stream. However, the only way I got this to work is by restarting the pipeline with a different configuration, which is very cumbersome since I need to take a lot of subsequent snapshots.

Is there a way to turn on/off the recording feature? Just like pressing the record/stop buttons in the Intel RealSense Viewer.

lramati, 2019-09-04:
Yes. Should look something like this:
```python
profile = pipe.start(config) # Store the recording profile used by the pipeline
auto recorder = profile.get_device().as_recorder() # Obtain the streaming device and cast it to recorder type
recorder.pause() # Pause recording while continuing to stream
recorder.resume() # Resume recording of running stream
```

bernland, 2019-09-05:
Thank you, @lramati!

In the meantime, I have found the save_single_frameset() method, which suits my needs perfectly as it allows for saving single frames with different file names. (Restarting the pipeline with a new config, on the contrary, takes quite some time.)

However, there are quite a few unanswered posts in regards to the depth frame not being recorded correctly: #4020, #2588 , #3671, and #3704.

Can you please provide an example showing how to correctly use the method?

lramati, 2019-09-05:
The save_single_frameset processing block neglected to save the value of depth_units associated with the frame. When querying this value from the bag, it would see that it didn't have one and default to 0, which resulted in functions like depth_frame.get_distance reporting a distance of 0, even though the depth data was correctly recorded.
A pull request to fix this is incoming, but since a Pre-release version just came out, it might be a while before this change makes it into the official releases and PuPi packages. However, you can compile the branch from source if you want a fix sooner

RealSenseCustomerSupport, 2019-10-25:
----------------------------------------------

@bernland With the suggestion of lramati, did you get it through now? Looking forward to your update. Thanks!

bernland, 2019-10-28:
Yes, @RealSenseCustomerSupport, since compiling the branch from source, the method save_single_frameset() has been working as expected. Thank you!

RealSenseCustomerSupport, 2019-11-04:
----------------------------------------------

@bernland Glad to know this resolved. Close it accordingly. Thanks!

closed 2019-11-04 by RealSenseCustomerSupport