realsenseai/librealsense #4802: Failed to update firmware using 'intel-realsense-dfu.exe'.

chechik, 2019-09-04:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 } | 
| Firmware Version                   | (05.11.06.250) | 
| Operating System & Version |   {Win 10  | 
| Kernel Version (Linux Only)    |  ----                                        | 
| Platform                                 | PC  |
| SDK Version                            |  {  2.<2>.<2> }                          | 
| Language                            |  {C#}                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
The attempt to update the firmware using 'intel-realsense-dfu.exe' tool, is failed with message:

_WaitForState(DFU_STATE::dfuDNLOAD_IDLE, 5000) Failed with status-2147467259
DownloadImage(reinterpret_cast<const BYTE *>(firmwareImage), imageLength) Failed with status-2147467259
Press any key to continue . . ._ 

The same result is by using _'3.Recover devices stuck in Upgrade Mode.'_ option (the camera has been converted to UpdateDevice using method UpdatableDevice.EnterUpdateState() of .Net API.)
The issue caused after several successful updates (less than 50 attempts) .

matkatz, 2019-09-04:
There is a limit of how many updates can be done to the same version, if you reached that limit the update procedure will fail and you will be able to update only to newer versions.

[Intel.RealSense.Viewer](https://github.com/IntelRealSense/librealsense/releases/download/v2.27.0/Intel.RealSense.Viewer.exe) now supports FW update and it should provide information in case of failure.

You can also try [rs-fw-update](https://github.com/IntelRealSense/librealsense/tree/master/tools/fw-update) tool if a console application is required.

matkatz, 2019-09-05:
Currently no.

closed 2019-09-10 by dorodnic