realsenseai/librealsense #4803: Using syncer along with pipeline

ahuizxc, 2019-09-04:
Hi 
Recently I encountered a breakdown about rs2::pipeline,
while I am using pipeline.start(callback) method, I want to get synced IMU data like accel and gyro at the same frameset. But I cannot find any information about this usage. What I find is all about using rs2::sensor and syncer.
so is there any sample code for this problem?
thanks a lot :)

MartyG-RealSense, 2019-09-04:
Intel have a guide about getting data from the IMU that has a sample script at the end of it in the section headed "API".  I hope this will be of use to you.

https://www.intelrealsense.com/how-to-getting-imu-data-from-d435i-and-t265/

I will post the listing here for your reading convenience.

`rs2::pipeline pipe;
rs2::config cfg;
cfg.enable_stream(RS2_STREAM_GYRO);
cfg.enable_stream(RS2_STREAM_ACCEL);
cfg.enable_stream(RS2_STREAM_POSE);
pipe.start(cfg);
while (app) // Application still alive?
{
    rs2::frameset frameset = pipe.wait_for_frames();
    // Find and retrieve IMU and/or tracking data
    if (rs2::motion_frame accel_frame = frameset.first_or_default(RS2_STREAM_ACCEL))
    {
        rs2_vector accel_sample = accel_frame.get_motion_data();
        //std::cout << "Accel:" << accel_sample.x << ", " << accel_sample.y << ", " << accel_sample.z << std::endl;
        //...
    }
    if (rs2::motion_frame gyro_frame = frameset.first_or_default(RS2_STREAM_GYRO))
    {
        rs2_vector gyro_sample = gyro_frame.get_motion_data();
        //std::cout << "Gyro:" << gyro_sample.x << ", " << gyro_sample.y << ", " << gyro_sample.z << std::endl;
        //...
    }
    if (rs2::pose_frame pose_frame = frameset.first_or_default(RS2_STREAM_POSE))
    {
        rs2_pose pose_sample = pose_frame.get_pose_data();
        //std::cout << "Pose:" << pose_sample.translation.x << ", " << pose_sample.translation.y << ", " << pose_sample.translation.z << std::endl;
        //...
    }
}`

ahuizxc, 2019-09-04:
I have already read this code before, but what I really want is something like "imu_stream" that can provide the accel and gyro data at the same time.

MartyG-RealSense, 2019-09-04:
I recall there was a discussion earlier this year about how to get image data and IMU data working together.

https://github.com/IntelRealSense/librealsense/issues/4018

ev-mp, 2019-09-05:
@ahuizxc hello,
The IMU streams use different clocks with "non-harmonic" data rates: 200/400 for Gyro and 63/250 for Accel . It is not feasible to generate a fully-synchronized accel/gyro tuples.
The approach used in the SDK is allow `pipeline` to decide how to proceed each stream:
- For UVC streams (depth/ir/color/fisheye) pipe's callback will generate a frameset of all the available video streams at the expense of latency.
- For IMU streams the frames will trigger the callbacks immediately to minimize latency, as in comparison to video, getting frames in timely manner is critical. But this also means that accel and gyro frames will  arrive one at a time.

Note that if the flow cannot provide for the user's scenario then it is possible to implement a user-defined syncer with a more fine-tuned logic.

ev-mp, 2019-09-26:
@ahuizxc, I'm closing this as answered.
Feel free to reopen as needed

closed 2019-09-26 by ev-mp

jyzhong, 2020-04-18:
@ev-mp hi, I'm wandering if there has a config to get synced IMU data. Just like the version of ROS , by setting "unite_imu_method", I can get a same framerate about the gyro and acc.