realsenseai/librealsense #4804: TypeError for playback.seek() in python wrapper

soarwing52, 2019-09-04:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (10)   | 
| SDK Version                            |  { pyrealsense2  2.27.0.1067}                          | 
| Language                            |  {python}                          | 
| Segment			|  {others }                          | 

### Issue Description
I'm still trying the seek function in the python wrapper since February,

I have the basic lines set up reading a bag file
```
pipeline = rs.pipeline()
config = rs.config()
config.enable_all_streams()
config.enable_device_from_file(bag, True)
profile = pipeline.start(config)
device = profile.get_device()
playback = device.as_playback()
```
and I can use current_status,file_name,get_duration,get_position,pause, resume, set_real_time without any error

and while running in a while loop, I have this line
```
        k = cv2.waitKey(50)

        if k == 32:
            playback.seek(0)
```

but only when I want to use seek, it throws back 
```
Traceback (most recent call last):
  File "C:/Users/cyh/Documents/GitHub/decision/ss.py", line 35, in <module>
    playback.seek(0)
TypeError: seek(): incompatible function arguments. The following argument types are supported:
    1. (self: pyrealsense2.pyrealsense2.playback, Sets the playback to a specified time point of the played data.time: datetime.timedelta) -> None

Invoked with: <pyrealsense2.device: Intel RealSense D435 (S/N: 817512071792)>, 0
```

but as I check my type of playback is :
<class 'pyrealsense2.pyrealsense2.playback'>
and I can use other functions of playback

so I wonder which part is wrong of this part

made a new Issue as @lramati said in #1545

lramati, 2019-09-04:
playback.seek() only works with floats and timedelta objects. the literal `0` is an int, and the translation layer doesn't do the two step process of promoting to float and then converting to timedelta. Using `0.` or `datetime.timedelta(0)` work for me.

lramati, 2019-09-04:
There's also an unfortunate glitch with the documentation where the docstring is being interpreted as part of the name of the second argument, effectively hiding it from the displayed signature. I'm looking to address this for the next release

soarwing52, 2019-09-05:
Thank you, timedelta(0) worked, and also using any float works.
But it seems like timestamp can't be used in this case?

lramati, 2019-09-08:
Unfortunately, no. The timestamps returned from withing librealsense are expressed as nanoseconds, and the python timedelta object is only exact to the microsecond, so we opted to leave the return values as integers rather than lose precision.

soarwing52, 2019-09-09:
@lramati  thank you for the explaination. 
I suppose the second can be converted either from frame numer with the correcsponding fps.

I will keep on optimizing my application, thank you!

closed 2019-09-09 by soarwing52