realsenseai/librealsense #4806: RealSense Viewer can't be opened in Win10

wanghelin1024, 2019-09-05:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
Hello,
        After I installed the latest full SDK, I can see the RealSense Viewer icon on my desktop. However, when I double click it , there is nothing happened , the softeware is not run. I tried some other versions of SDK and use my teamates's laptop have a test. They are all the same issue. Looking forward to your feedback, many thanks.

BTW; From my running program, the SDK works fine, it means SDK installed successfully

github path:
Intel.RealSense.SDK.exe
https://github.com/IntelRealSense/librealsense/releases/tag/v2.27.0

ev-mp, 2019-09-05:
@wanghelin1024 hello,
The `Realsense-viewer` uses Windows-predefined user settings to store application-specific settings and logs: " C:\users\<username>\Documents".
We have received similar reports by users whose Document's path included either non-utf strings or was a network-mapped path. #4569
Can you check whether this applies in your case?

wanghelin1024, 2019-09-06:
> @wanghelin1024 hello,
> The Realsense-viewer uses Windows-predefined user settings to store application-specific settings and logs: " C:\users<username>\Documents".
> We have received similar reports by [...]

Thanks for your quick support !!!
I checked my Document's path (C:\Users\A11190321040347\Documents), I think it should be OK.
Maybe it is not the same issue as #4569,  do you have some other suggestions? Many thanks.

gycka, 2019-09-06:
I had this problem when the RealSense Viewer not opening at all. I found this to be caused by the disabled graphics card, specifically it was for the Intel NUC Hades Canyon. 

The NUC was operated headless, but it stops working with enabled graphics card (black screen) until a monitor was attached. Disabled graphics did not allow to view RealSense viewer, enabled graphics did not allow to remotely view the screen. I installed a ghost monitor plug and now everything is working. In any case, your problem might be different, but check if you have a graphics card.

Dontla, 2019-09-10:
wo mei you yu dao guo ni zhe yang de qing kuang , ni ke yi chang shi you jian yi guan li yuan shen fen yun xing, ru guo zai bu xing de hua ke yi chang shi guan bi sha du ruan jian, huo zhe guan bi xi tong de fang huo qiang!

wanghelin1024, 2019-09-11:
> I had this problem when the RealSense Viewer not opening at all. I found this to be caused by the disabled graphics card, specifically it was for the Intel NUC Hades Canyon.
> The NUC was operated [...]

Thanks a lot for your suggestion~
After I re-installed the driver of graphics card, the RealSense Viewer works fine

wanghelin1024, 2019-09-11:
> wo mei you yu dao guo ni zhe yang de qing kuang , ni ke yi chang shi you jian yi guan li yuan shen fen yun xing, ru guo zai bu xing de hua ke yi chang shi guan bi sha du ruan jian, huo zhe guan bi xi tong de fang huo qiang!

Thanks for your help~
This issue has been resolved, it is caused by the disabled graphics card.

wanghelin1024, 2019-09-11:
Thanks for all your supports, I will close this issue.

closed 2019-09-11 by wanghelin1024