realsenseai/librealsense #4807: Python API for A-factor?

keithahern, 2019-09-05:
Hello,

Is there python bindings for the new A-factor functionality?

From github it looks like:
C
rs2_set_amp_factor
rs2_get_amp_factor

The white paper mentions
C
RS2_Set_A-factor
RS2_Get_A-factor
C++
set_new_depth_control
get_new_depth_control

Thanks

lramati, 2019-09-05:
try rs.advanced_mode.get/set_amp_factor

keithahern, 2019-09-05:
Thank you, I've got a little closer but not there yet.

Tried this:
```
advnc_mode = rs.rs400_advanced_mode(self.profile.get_device())
advnc_mode.set_amp_factor(0.08)
```

got this:
```
TypeError: set_amp_factor(): incompatible function arguments. The following argument types are supported:
    1. (self: pyrealsense2.rs400_advanced_mode, group: STAFactor) -> None
```

Looking at some other examples I tried the following:

```
advnc_mode = rs.rs400_advanced_mode(self.profile.get_device())
current_sta_factor_group = advnc_mode.get_amp_factor()
current_sta_factor_group.amplitude_factor = 0.08
advnc_mode.set_amp_factor(current_sta_factor_group)
```

But got the following
```
current_sta_factor_group = advnc_mode.get_amp_factor()
TypeError: Unable to convert function return value to a Python type! The signature was
        (self: pyrealsense2.rs400_advanced_mode, mode: int=0) -> STAFactor

```

lramati, 2019-09-05:
There was an unfortunate oversight when adding the functions to the wrapper. I will open a pull request soon to address the issue, but as a pre-release version just went out it might take some time before it makes it into the official release and PyPi packages. You will be able to compile the branch from source if you'd like a fix sooner

keithahern, 2019-09-05:
I’m compiling from source anyway.

Thank you Lior I’ll look out for the fix.

keithahern, 2019-09-05:
I've checked out the pull request, compiled the SDK, and I'm no longer getting the error, will continue testing to confirm it's working.

thank you.

RealSenseCustomerSupport, 2019-10-09:
----------------------------------------------

The pull request was integrated since 2.28.1.

closed 2019-10-16 by RealSenseCustomerSupport

Microset, 2019-11-16:
Good day!
Librealsense SDK 2.30
Try to read and write A_factor:

advnc_mode = rs.rs400_advanced_mode(self.profile.get_device())
advnc_mode.get_amp_factor()
print(advnc_mode.get_amp_factor())
advnc_mode.set_amp_factor(0.08)

TypeError: set_amp_factor(): incompatible function arguments. The following argument types are supported:
    1. (self: pyrealsense2.rs400_advanced_mode, group: STAFactor) -> None

Can read but not write A_factor.
Please check in your side.

Best rerads