realsenseai/librealsense #4810: Announcement: FRAMOS D435e RealSense industrial camera is now available to order

MartyG-RealSense, 2019-09-05:
Hi everyone,

Recently I highlighted a new industrial own-brand version of the RealSense D435 by the Intel Authorized Distributor company FRAMOS, called the FRAMOS D435e.

https://github.com/IntelRealSense/librealsense/issues/4769

The D435e is now available to order in the FRAMOS store.  Its industrial nature means that there is a price premium over the standard Intel RealSense D435 model, though it may provide industrial and OEM users with options only otherwise available through custom build projects.

https://shop-us.framos.com/FRAMOS-Industrial-Depth-Camera-D435e-Starter-Kit-p147712085

ShettyHarapanahalli, 2019-09-24:
What is the cost of this device?

MartyG-RealSense, 2019-09-24:
It is normally $1299 USD but is on sale for $1099 at the time of writing this.

https://shop-us.framos.com/FRAMOS-Industrial-Depth-Camera-D435e-Starter-Kit-p147712085

closed 2019-10-04 by MartyG-RealSense