realsenseai/librealsense #4811: roslaunch rs_camera.launch fails for a D415

evarcurto, 2019-09-06:
Hi, I'm trying to connect my D415 camera but when I run `roslaunch realsense2_camera rs_camera.launch`  the following error appears: 
[ERROR] [1567766150.564553987]: Region-of-interest is not implemented for this device!
I'm using Ubuntu 16.04, kernel version 4.15.0-60-generic. My firmware version is 05.10.06.00.
[image]
[image]

hongluck, 2019-09-06:
same in D435

RealSenseCustomerSupport, 2019-10-02:
----------------------------------------------

Hi @evarcurto,

Please update the Firmware on your camera to the latest version and try to run the same command.
The latest version, as of today, of Firmware for Intel RealSense D400 Product Family is 5.11.15.0 and it can be found here:
https://downloadcenter.intel.com/download/29078/Latest-Firmware-for-Intel-RealSense-D400-Product-Family?product=128256

Thank you!

RealSenseCustomerSupport, 2019-10-22:
----------------------------------------------

Hi @evarcurto,

Do you need any further clarifications regarding this topic?

closed 2019-10-29 by RealSenseCustomerSupport