realsenseai/librealsense #4812: Can't build android library

Murded, 2019-09-06:
Hi, I've followed the tutorial shown at [https://github.com/IntelRealSense/librealsense/tree/master/wrappers/android](url) on how to build the android library, however when I execture the assembleRelease task I got the following output (I'm using Windows 10 - Android studio 3.4.1).

```
12:53:13: Executing task 'assembleRelease'...

> Task :librealsense:preBuild UP-TO-DATE
> Task :librealsense:preReleaseBuild UP-TO-DATE
> Task :librealsense:compileReleaseAidl NO-SOURCE
> Task :librealsense:compileReleaseRenderscript UP-TO-DATE
> Task :librealsense:checkReleaseManifest UP-TO-DATE
> Task :librealsense:generateReleaseBuildConfig UP-TO-DATE
> Task :librealsense:generateReleaseResValues UP-TO-DATE
> Task :librealsense:generateReleaseResources UP-TO-DATE
> Task :librealsense:packageReleaseResources UP-TO-DATE
> Task :librealsense:processReleaseManifest UP-TO-DATE
> Task :librealsense:generateReleaseRFile UP-TO-DATE
> Task :librealsense:prepareLintJar UP-TO-DATE
> Task :librealsense:generateReleaseSources UP-TO-DATE
> Task :librealsense:javaPreCompileRelease UP-TO-DATE
> Task :librealsense:compileReleaseJavaWithJavac UP-TO-DATE
> Task :librealsense:extractReleaseAnnotations
> Task :librealsense:mergeReleaseConsumerProguardFiles UP-TO-DATE
> Task :librealsense:mergeReleaseShaders UP-TO-DATE
> Task :librealsense:compileReleaseShaders UP-TO-DATE
> Task :librealsense:generateReleaseAssets UP-TO-DATE
> Task :librealsense:packageReleaseAssets UP-TO-DATE
> Task :librealsense:packageReleaseRenderscript NO-SOURCE
> Task :librealsense:processReleaseJavaRes NO-SOURCE
> Task :librealsense:transformResourcesWithMergeJavaResForRelease UP-TO-DATE
> Task :librealsense:transformClassesAndResourcesWithSyncLibJarsForRelease UP-TO-DATE

> Task :librealsense:generateJsonModelRelease
CMake Error at C:/Users/Andrew/AppData/Local/Android/sdk/cmake/3.10.2.4988404/share/cmake-3.10/Modules/FindX11.cmake:429 (message):
  Could not find X11
Call Stack (most recent call first):
  ../../../third-party/glfw/CMakeLists.txt:220 (find_package)

CMake Error at C:/Users/Andrew/AppData/Local/Android/sdk/cmake/3.10.2.4988404/share/cmake-3.10/Modules/FindX11.cmake:429 (message):
  Could not find X11
Call Stack (most recent call first):
  ../../../third-party/glfw/CMakeLists.txt:220 (find_package)

Configuration failed.
CMake Error at C:/Users/Andrew/AppData/Local/Android/sdk/cmake/3.10.2.4988404/share/cmake-3.10/Modules/FindX11.cmake:429 (message):
  Could not find X11
Call Stack (most recent call first):
  ../../../third-party/glfw/CMakeLists.txt:220 (find_package)

CMake Error at C:/Users/Andrew/AppData/Local/Android/sdk/cmake/3.10.2.4988404/share/cmake-3.10/Modules/FindX11.cmake:429 (message):
  Could not find X11
Call Stack (most recent call first):
  ../../../third-party/glfw/CMakeLists.txt:220 (find_package)

Configuration failed.
CMake Warning at D:\librealsense\wrappers\android\librealsense\..\..\..\CMakeLists.txt:10 (message):
  No internet connection, disabling BUILD_WITH_TM2

CMake Warning at D:\librealsense\wrappers\android\librealsense\..\..\..\CMakeLists.txt:10 (message):
  No internet connection, disabling BUILD_WITH_TM2

CMake Warning at D:\librealsense\wrappers\android\librealsense\..\..\..\CMakeLists.txt:16 (message):
  No internet connection, disabling IMPORT_DEPTH_CAM_FW

CMake Warning at D:\librealsense\wrappers\android\librealsense\..\..\..\CMakeLists.txt:16 (message):
  No internet connection, disabling IMPORT_DEPTH_CAM_FW

CMake Error at C:/Users/Andrew/AppData/Local/Android/sdk/cmake/3.10.2.4988404/share/cmake-3.10/Modules/FindX11.cmake:429 (message):
  Could not find X11
Call Stack (most recent call first):
  ../../../third-party/glfw/CMakeLists.txt:220 (find_package)
[... 125 lines cut ...]
Performing Test COMPILER_SUPPORTS_CXX0X
Performing Test COMPILER_SUPPORTS_CXX0X - Success
Setting Unix configurations
using RS2_USE_V4L2_BACKEND
libusb not found; using internal version (missing: LIBUSB_LIB LIBUSB_INC) 
GLFW 3.3 not found; using internal version
Looking for pthread.h
Looking for pthread.h - found
Looking for pthread_create
Looking for pthread_create - found
Found Threads: TRUE  
Could NOT find Vulkan (missing: VULKAN_LIBRARY VULKAN_INCLUDE_DIR) 
Using X11 for window creation
CMake Error at C:/Users/Andrew/AppData/Local/Android/sdk/cmake/3.10.2.4988404/share/cmake-3.10/Modules/FindX11.cmake:429 (message):
  Could not find X11
Call Stack (most recent call first):
  ../../../third-party/glfw/CMakeLists.txt:220 (find_package)

Configuring incomplete, errors occurred!
See also "D:/librealsense/wrappers/android/librealsense/.externalNativeBuild/cmake/release/armeabi/CMakeFiles/CMakeOutput.log".

12:54:05: Task execution finished 'assembleRelease'.
```

Any suggestions on what I could be missing?

closed 2019-09-06 by Murded

Virinchi2595, 2019-12-12:
Hi 
I am also facing this issue
any solution to this?

Murded, 2019-12-12:
> Hi
> I am also facing this issue
> any solution to this?

I wasn't able to get it working on the machine I was originally trying on but I tried it on a different machine and it just built so I'm not sure what the issue was in the end

Virinchi2595, 2019-12-12:
Can you please tell me the configuration of machine you used to build?

Virinchi2595, 2019-12-12:
> > Hi
> > I am also facing this issue
> > any solution to this?
> 
> I wasn't able to get it working on the machine I was originally trying on but I tried it on a different machine and it just built so I'm not sure what the issue was in the end

Can you please tell me the configuration of machine you used to build?

Jiloc, 2020-01-13:
for future reference: `apt-get install libx11-dev xorg-dev libglu1-mesa-dev`