realsenseai/librealsense #4813: vertex normal information is not present in ply file generated from bag file

rupeshchandgude, 2019-09-06:
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | RealSense Viewer  | 
| Operating System & Version | Ubuntu 16 | 
| Platform                                 | PC |
| Language                            | python | 

### Issue Description
I have generated bag file using Realsense Viewer. After I convert it into ply file using following code:

> pc = rs.pointcloud()
points = rs.points() 
pipeline = rs.pipeline()
config = rs.config()
rs.config.enable_device_from_file(config,bag_file,repeat_playback=False)
pipeline.start(config)
colorizer = rs.colorizer()
align_to = rs.stream.color
align = rs.align(align_to)
frames = pipeline.wait_for_frames()
frames = align.process(frames)
depth_frame = frames.get_depth_frame()
color = frames.get_color_frame()
depth_color_frame = colorizer.colorize(depth_frame)
depth_color_image = np.asanyarray(depth_color_frame.get_data())
pc.map_to(depth_color_frame)
points = pc.calculate(depth_frame)
points.export_to_ply('temp.ply',color)

Generated ply file does not have vertex normal information in it. Is there any way to add vertex normal information while generating ply file or after generating ply file?

lramati, 2019-09-08:
You would have to turn the pointcloud into a mesh and then calculate vertex normals based on that mesh. I think PCL gets recommended frequently by people here for doing these kinds of things, although that appears to be a C++ project

rupeshchandgude, 2019-09-09:
thanks @lramati  I will try that out.

dorodnic, 2019-12-10:
@rupeshchandgude - latest viewer includes options to export with normals. Thank you for the feedback

closed 2019-12-10 by dorodnic

rupeshchandgude, 2019-12-17:
@dorodnic  Thanks for update.

djflstkddk, 2020-04-03:
@dorodnic 
Hi there.  I have the same issue, but in my case, I should do it in python code, not realsense-viewer. Is there a way to get .ply file with both rgb color & vertex normals with python code? 
my issue is in #6194. I guess u know how to do it maybe.. 
thanks in advance.

lramati, 2020-04-12:
check out the save_to_ply object