realsenseai/librealsense #4815: No realsense string observed in install

hongluck, 2019-09-06:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |    Linux (Ubuntu 16.04)   | 
| Kernel Version (Linux Only)    |   4.15.0-60                                         | 
| Platform                                 | PC|
| SDK Version                            |                        | 
| Language                            |  {C/C#/}                          | 
| Segment			|  {Robot }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

In librealsense ubuntu 16.04 installation git, there is a command to check if the realsense string is in. When I run it it does not show any string including realsense.
I am having problems connecting to my realsense due to W10 problem

$ modinfo uvcvideo | grep "version:"
version:        1.1.1
srcversion:     79A49299483C7A964CCC049
This is the link for it I will copy and paste the issue as well
https://github.com/IntelRealSense/librealsense/issues/4794#issuecomment-528487804

Hello I hava a D415 Depth Camera and it won't connect to my laptop.
The main error for this is mentioned below.
I can access the realsense viewer without any problems but when ever I turn my node on by using
roslaunch realsense2_camera rs_camera.launch
it gives this error. Please help me thank you

My kernel version is 4.15.0-60-generic

An exception has been thrown: The requested pixel format 'W10 ' is not natively supported by the Linux kernel and likely requires a patch!
Alternatively please upgrade to kernel 4.12 or later. Last Error: Invalid argument
terminate called after throwing an instance of 'rs2::backend_error'
what(): The requested pixel format 'W10 ' is not natively supported by the Linux kernel and likely requires a patch!
Alternatively please upgrade to kernel 4.12 or later. Last Error: Invalid argument
[camera/realsense2_camera_manager-1] process has died [pid 15139, exit code -6, cmd /opt/ros/kinetic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/hong/.ros/log/68c59866-ce8c-11e9-8e37-a86daaec59a0/camera-realsense2_camera_manager-1.log].
log file: /home/hong/.ros/log/68c59866-ce8c-11e9-8e37-a86daaec59a0/camera-realsense2_camera_manager-1*.log

ev-mp, 2019-09-09:
@hongluck , there was a breaking kernel change introduced in 4.15.0-58 which broke the DKMS for that branch. We're working on an update.
In the meanwhile for kernels  4.15.0-58+ please use a DKMS package attached in #4669

hongluck, 2019-09-09:
Still giving me the error as the link
https://github.com/IntelRealSense/librealsense/issues/4794#issuecomment-528487804

[ERROR] [1568036482.236293983]: An exception has been thrown: The requested pixel format 'W10 ' is not natively supported by the Linux kernel and likely requires a patch!
Alternatively please upgrade to kernel 4.12 or later. Last Error: Invalid argument
terminate called after throwing an instance of 'rs2::backend_error'
  what():  The requested pixel format 'W10 ' is not natively supported by the Linux kernel and likely requires a patch!
Alternatively please upgrade to kernel 4.12 or later. Last Error: Invalid argument
[camera/realsense2_camera_manager-2] process has died [pid 21254, exit code -6, cmd /opt/ros/kinetic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/hong/.ros/log/81a4ec90-d307-11e9-83f4-a86daaec59a0/camera-realsense2_camera_manager-2.log].
log file: /home/hong/.ros/log/81a4ec90-d307-11e9-83f4-a86daaec59a0/camera-realsense2_camera_manager-2*.log
^C[camera/realsense2_camera-3] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete

I dont think the deb file you gave made any changes

ev-mp, 2019-09-09:
Can you provide the full installation log of the deb ?

hongluck, 2019-09-10:
Error log is this

roslaunch realsense2_camera rs_camera.launch
... logging to /home/hong/.ros/log/f84c8fd0-d3d2-11e9-b448-a86daaec59a0/roslaunch-hong-JDL-HELIOS-11324.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://hong-JDL-HELIOS:40281/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: True
 * /camera/realsense2_camera/enable_fisheye2: True
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 400
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra_fps: 30
 * /camera/realsense2_camera/infra_height: 480
 * /camera/realsense2_camera/infra_width: 640
[... 105 lines cut ...]
mod: videodev 			
mod: uvcvideo 			 version: 1.1.1
mod: hid_sensor_gyro_3d 	
mod: hid_sensor_accel_3d 	
Setting up librealsense2-dkms (1.3.5-0ubuntu1) ...
Loading new librealsense2-dkms-1.3.5 DKMS files...
Building only for 4.15.0-60-generic
Building initial module for 4.15.0-60-generic
ERROR: Cannot create report: [Errno 17] File exists: '/var/crash/librealsense2-dkms.0.crash'
Error! Bad return status for module build on kernel: 4.15.0-60-generic (x86_64)
Consult /var/lib/dkms/librealsense2-dkms/1.3.5/build/make.log for more information.

Loading the modified modules into kernel... complete
Current status:
bbswitch, 0.8, 4.15.0-45-generic, x86_64: installed
bbswitch, 0.8, 4.15.0-60-generic, x86_64: installed
bbswitch, 0.8, 4.4.0-161-generic, x86_64: installed
librealsense2-dkms, 1.3.5: added
nvidia-430, 430.26, 4.15.0-45-generic, x86_64: installed
nvidia-430, 430.26, 4.15.0-60-generic, x86_64: installed
nvidia-430, 430.26, 4.4.0-161-generic, x86_64: installed
mod: videodev 			
mod: uvcvideo 			 version: 1.1.1
mod: hid_sensor_gyro_3d 	
mod: hid_sensor_accel_3d

hongluck, 2019-09-10:
This is the make error shown above.
ERROR: Cannot create report: [Errno 17] File exists: '/var/crash/librealsense2-dkms.0.crash'
Error! Bad return status for module build on kernel: 4.15.0-60-generic (x86_64)
Consult /var/lib/dkms/librealsense2-dkms/1.3.5/build/make.log for more information.

Below is the make.log file

DKMS make.log for librealsense2-dkms-1.3.5 for kernel 4.15.0-60-generic (x86_64)
Tue Sep 10 10:00:58 EDT 2019
make: Entering directory '/usr/src/linux-headers-4.15.0-60-generic'
  CC [M]  /var/lib/dkms/librealsense2-dkms/1.3.5/build/4.15.0/drivers/media/usb/uvc/uvc_driver.o
In file included from ./arch/x86/include/asm/atomic.h:5:0,
                 from ./include/linux/atomic.h:5,
                 from /var/lib/dkms/librealsense2-dkms/1.3.5/build/4.15.0/drivers/media/usb/uvc/uvc_driver.c:14:
./arch/x86/include/asm/qspinlock.h: In function ‘native_queued_spin_unlock’:
./arch/x86/include/asm/qspinlock.h:42:25: error: ‘struct qspinlock’ has no member named ‘locked’
  smp_store_release(&lock->locked, 0);
                         ^
./include/linux/compiler.h:304:9: note: in definition of macro ‘__compiletime_assert’
   if (!(condition))     \
         ^
./include/linux/compiler.h:324:2: note: in expansion of macro ‘_compiletime_assert’
  _compiletime_assert(condition, msg, __compiletime_assert_, __LINE__)
  ^
./include/linux/compiler.h:327:2: note: in expansion of macro ‘compiletime_assert’
  compiletime_assert(__native_word(t),    \
  ^
./include/linux/compiler.h:327:21: note: in expansion of macro ‘__native_word’
  compiletime_assert(__native_word(t),    \
                     ^
./arch/x86/include/asm/barrier.h:97:2: note: in expansion of macro ‘compiletime_assert_atomic_type’
  compiletime_assert_atomic_type(*p);    \
  ^
/var/lib/dkms/librealsense2-dkms/1.3.5/build/4.15.0/drivers/media/usb/uvc/../../../../include-overrides/asm-generic/barrier.h:157:33: note: in expansion of macro ‘__smp_store_release’
 #define smp_store_release(p, v) __smp_store_release(p, v)
                                 ^
./arch/x86/include/asm/qspinlock.h:42:2: note: in expansion of macro ‘smp_store_release’
  smp_store_release(&lock->locked, 0);
  ^
./arch/x86/include/asm/qspinlock.h:42:25: error: ‘struct qspinlock’ has no member named ‘locked’
  smp_store_release(&lock->locked, 0);
                         ^
./include/linux/compiler.h:304:9: note: in definition of macro ‘__compiletime_assert’
   if (!(condition))     \
         ^
./include/linux/compiler.h:324:2: note: in expansion of macro ‘_compiletime_assert’
  _compiletime_assert(condition, msg, __compiletime_assert_, __LINE__)
  ^
./include/linux/compiler.h:327:2: note: in expansion of macro ‘compiletime_assert’
  compiletime_assert(__native_word(t),    \
  ^
./include/linux/compiler.h:327:21: note: in expansion of macro ‘__native_word’
  compiletime_assert(__native_word(t),    \
                     ^
./arch/x86/include/asm/barrier.h:97:2: note: in expansion of macro ‘compiletime_assert_atomic_type’
  compiletime_assert_atomic_type(*p);    \
  ^
/var/lib/dkms/librealsense2-dkms/1.3.5/build/4.15.0/drivers/media/usb/uvc/../../../../include-overrides/asm-generic/barrier.h:157:33: note: in expansion of macro ‘__smp_store_release’
 #define smp_store_release(p, v) __smp_store_release(p, v)
                                 ^
./arch/x86/include/asm/qspinlock.h:42:2: note: in expansion of macro ‘smp_store_release’
  smp_store_release(&lock->locked, 0);
  ^
./arch/x86/include/asm/qspinlock.h:42:25: error: ‘struct qspinlock’ has no member named ‘locked’
  smp_store_release(&lock->locked, 0);
                         ^
./include/linux/compiler.h:304:9: note: in definition of macro ‘__compiletime_assert’
   if (!(condition))     \
         ^
./include/linux/compiler.h:324:2: note: in expansion of macro ‘_compiletime_assert’
[... 82 lines cut ...]
                                          ^
/var/lib/dkms/librealsense2-dkms/1.3.5/build/4.15.0/drivers/media/usb/uvc/../../../../include-overrides/asm-generic/barrier.h:157:33: note: in expansion of macro ‘__smp_store_release’
 #define smp_store_release(p, v) __smp_store_release(p, v)
                                 ^
./arch/x86/include/asm/qspinlock.h:42:2: note: in expansion of macro ‘smp_store_release’
  smp_store_release(&lock->locked, 0);
  ^
In file included from ./include/linux/scatterlist.h:8:0,
                 from ./include/linux/spi/spi.h:23,
                 from /var/lib/dkms/librealsense2-dkms/1.3.5/build/4.15.0/drivers/media/usb/uvc/../../../../include-overrides/media/v4l2-common.h:195,
                 from /var/lib/dkms/librealsense2-dkms/1.3.5/build/4.15.0/drivers/media/usb/uvc/uvc_driver.c:26:
./include/linux/mm.h: In function ‘mm_inc_nr_puds’:
./include/linux/mm.h:1685:6: error: implicit declaration of function ‘mm_pud_folded’ [-Werror=implicit-function-declaration]
  if (mm_pud_folded(mm))
      ^
./include/linux/mm.h: In function ‘mm_inc_nr_pmds’:
./include/linux/mm.h:1713:6: error: implicit declaration of function ‘mm_pmd_folded’ [-Werror=implicit-function-declaration]
  if (mm_pmd_folded(mm))
      ^
cc1: some warnings being treated as errors
scripts/Makefile.build:330: recipe for target '/var/lib/dkms/librealsense2-dkms/1.3.5/build/4.15.0/drivers/media/usb/uvc/uvc_driver.o' failed
make[1]: *** [/var/lib/dkms/librealsense2-dkms/1.3.5/build/4.15.0/drivers/media/usb/uvc/uvc_driver.o] Error 1
Makefile:1747: recipe for target 'uvcvideo.ko' failed
make: *** [uvcvideo.ko] Error 2
make: Leaving directory '/usr/src/linux-headers-4.15.0-60-generic'

fgrzadkowski, 2019-09-13:
@ev-mp I'm seeing the same error also for an older kernel - `4.4.0-131-generic`.

Info:
* SDK version 2.25.0-0~realsense0.1362
* Realsense ROS version 2.2.8
* Ubuntu 16.04
* Kernel: 4.4.0-131-generic
* Nvidia driver version: 390.48 
* DKMS version: 1.3.4-0ubuntu1

Any advice how to workaround the problem?

ev-mp, 2019-09-15:
@fgrzadkowski , 
I've checked SDK 2.25.0 with the kernel version 4.4.0-131-generic and the SDK works properly with no obvious misbehaviors
Some observations:
- The message regarding missing `W10` descriptor is a warning that can be disregarded - it doesn't affect functionality as this format is not supported by D415/D435 models. And it was actually silenced in later SDK versions. 
- The DKMS version you're using is not up-to-date. The recommended one is 1.3.5.
- Since you're running NVIDIA video driver, which itself is heavily based on DKMS - you should check that installing Librealsense DKMS package goes smooth. You can monitor the installation of `sudo apt install librealsense2-dkms` with a kernel log opened in another console `dmesg -wT.

With all that said it is not clear what is the issue that you've encountered, so pls attach the exact error for evaluation. Also try to reproduce the issues with SDK demos, such as `realsense-viewer` to further isolate the issue

fgrzadkowski, 2019-09-16:
This is the log message that I have:
```
A 2019-09-13T07:56:02.296917635Z  13/09 07:56:02,296 WARNING [140034679183104] (backend-v4l2.cpp:1149) Pixel format 20303157-0000-0010-8000-00aa003 likely requires patch for fourcc code W10 ! 
A 2019-09-13T07:56:02.298548176Z  13/09 07:56:02,298 ERROR [140034679183104] (types.h:243) The requested pixel format 'W10 ' is not natively supported by the Linux kernel and likely requires a patch! 
A 2019-09-13T07:56:02.298550162Z Alternatively please upgrade to kernel 4.12 or later. Last Error: Invalid argument 
A 2019-09-13T07:56:02.299257939Z  [31m[ERROR] [1568361362.299137722]: An exception has been thrown: The requested pixel format 'W10 ' is not natively supported by the Linux kernel and likely requires a patch! 
A 2019-09-13T07:56:02.299259868Z Alternatively please upgrade to kernel 4.12 or later. Last Error: Invalid argument [0m 
A 2019-09-13T07:56:02.299416264Z terminate called after throwing an instance of 'rs2::backend_error' 
A 2019-09-13T07:56:02.299436137Z   what():  The requested pixel format 'W10 ' is not natively supported by the Linux kernel and likely requires a patch! 
A 2019-09-13T07:56:02.299436648Z Alternatively please upgrade to kernel 4.12 or later. Last Error: Invalid argument 
A 2019-09-13T07:56:02.486309146Z  [31m[camera_cart_left/realsense2_camera_manager-1] process has died [pid 104, exit code -6, cmd /opt/ros/kinetic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/root/.ros/log/bbbf4e56-d5fb-11e9-b5e1-a0369f04326f/camera_cart_left-realsense2_camera_manager-1.log]. 
A 2019-09-13T07:56:02.486311496Z log file: /root/.ros/log/bbbf4e56-d5fb-11e9-b5e1-a0369f04326f/camera_cart_left-realsense2_camera_manager-1*.log [0m 
A 2019-09-13T07:56:02.975468492Z  [1mprocess[box_placing_depth_baseline_srv-1]: started with pid [438] [0m 
A 2019-09-13T07:56:02.976452294Z  [1mprocess[camera_calibration_downloader_eddie_333_1560272407429094588-1]: started with pid [444] [0m 
A 2019-09-13T07:56:02.976766641Z  [1mprocess[robot_state_publisher-1]: started with pid [447] [0m 
A 2019-09-13T07:56:03.077657230Z shutting down processing monitor... 
A 2019-09-13T07:56:03.077775899Z ... shutting down processing monitor complete 
A 2019-09-13T07:56:03.077818606Z  [1mdone [0m 
A 2019-09-13T07:56:03.200225446Z  [34mSuccessful exit from subprocess: ['roslaunch', 'nomagic_launch', 'camera_calibration_downloader.launch', 'camera_id:=box_placing', 'allow_no_internet:=False', 'camera_serial_no:=810512061758'] [0m 
```

Also, from the code it looks like a regular exception:
https://github.com/IntelRealSense/librealsense/blob/4f2eac248530ef75fb38b95002522aa032165638/src/linux/backend-v4l2.cpp#L663

Am I mising something? @ev-mp

fgrzadkowski, 2019-09-16:
@ev-mp  It turned out to be a problem with not properly installed DKMS for the driver. Thanks!

ev-mp, 2019-09-18:
@fgrzadkowski , thanks for the confirmation.
Also, as noted by @doronhi in that thread the proper solution was [already integrated](https://github.com/IntelRealSense/realsense-ros/pull/924) into the ROS repository so the new version will run correctly even on "unpatched" Ubuntu machines.

@hongluck, to avoid the error please install DKMS or run the patches script. #4794

ev-mp, 2019-09-26:
@hongluck, did you manage to resolve it as suggested?
We'll close this if no further questions.

closed 2019-10-12 by RealSenseCustomerSupport