realsenseai/librealsense #4816: How to use the CUDA implementation for the alignment process?

kgh475926, 2019-09-07:
Camera: SR300
OS: Ubuntu
Platform: PC
Language: C++

What are the steps needed to use the CUDA implementation for the alignment process?
I tried to search any docs that can get me started but I haven't found any.

My understanding for CUDA so far is compile with the  `cmake ../ -DBUILD_WITH_CUDA=true` but I do not know what to do further.

Before recognizing the existence of CUDA, I was using the SSE implementation and I often referred to the sse-align.cpp with the` -DBUILD_EXAMPLES=true` but I'm wondering if there are any examples for the CUDA implementation.

Could I get some help?
Also, what is Jetson?

dorodnic, 2019-09-08:
For CUDA installation you need to follow official NVidia documentation. If you build the SDK with CMake flag `-DBUILD_WITH_CUDA=true` and the compilation (`make && sudo make install`)  completed without errors, your version of the SDK is using CUDA (no extra steps needed)
Similar to `sse-align.cpp` you have [`cuda-align.cu`](https://github.com/IntelRealSense/librealsense/blob/master/src/proc/cuda/cuda-align.cu) with all the implementation details

kgh475926, 2019-09-08:
@dorodnic I cmake-ed with `-DBUILD_WITH_CUDA=true` and also `make && sudo make install` but when it comes to the align object construction, the program keeps calling the align_sse instead of align_cuda.
Is does this mean that CUDA doesn't work for alignment?

RealSenseCustomerSupport, 2019-10-07:
----------------------------------------------

Hi @kgh475926,
Is this still a problem for you?
I was able to confirm that the the align_cuda works on both Xavier and PC with CUDA capable GPU.  For reference, here is my system information on the PC:

    :~$ lspci | grep -i nvidia
    01:00.0 VGA compatible controller: NVIDIA Corporation GP104 [GeForce GTX 1080] (rev a1)
    01:00.1 Audio device: NVIDIA Corporation GP104 High Definition Audio Controller (rev a1)

    :~$ uname -m && cat /etc/*release
    x86_64
    DISTRIB_ID=Ubuntu
    DISTRIB_RELEASE=18.04
    DISTRIB_CODENAME=bionic
    DISTRIB_DESCRIPTION="Ubuntu 18.04.3 LTS"

    :~$ cuda-gdb --version
    NVIDIA (R) CUDA Debugger
    10.1 release
    Portions Copyright (C) 2007-2019 NVIDIA Corporation
    GNU gdb (GDB) 7.12

Please let us know if you need further assistance. Thank you.

closed 2019-10-15 by RealSenseCustomerSupport