realsenseai/librealsense #4817: How to switch from __SSSE3__ to RS2_USE_CUDA?

kgh475926, 2019-09-07:
C++, Ubuntu, PC

How can one switch from __SSSE3__ to RS2_USE_CUDA?

In a file called processing-blocks-factory.cpp, the function std::make_shared<librealsense::align_sse>(align_to) gets invoked because some macro is set to __SSSE3__ but within the same file, I want to invoke std::make_shared<librealsense::align_cuda>(align_to) which is only invoked for the macro RS2_USE_CUDA.

Is there a way to direct the program to execute only RS2_USE_CUDA macros?

RealSenseCustomerSupport, 2019-10-15:
----------------------------------------------

Hi @kgh475926,

Do you enable "BUILD_WITH_CUDA" option in the cmake configuration?

RealSenseCustomerSupport, 2019-11-01:
----------------------------------------------

Hi @kgh475926,

Any update? Is the issue clarified/resolved?

RealSenseCustomerSupport, 2019-12-05:
----------------------------------------------

If no additional question, will close this.

closed 2019-12-05 by RealSenseCustomerSupport