realsenseai/librealsense #4818: Kernel 4.19 support for Raspberry Pi 4

edgarcamilocamacho, 2019-09-07:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 / T265 | 
| Firmware Version                   | 05.11.11.100 / 0.1.0.279 | 
| Operating System & Version |   Linux (Raspbian Buster)  | 
| Kernel Version (Linux Only)    |  4.19.69 | 
| Platform                                 | Raspberry Pi 4  |
| SDK Version                            | librealsense2  | 
| Language                            | ROS | 
| Segment			| Robot | 

### Issue Description
Are you planning to add support to Kernel 4.19? In order to use the cameras in Raspberry 4 and Raspbian Buster.

I am getting the error of this issue: https://github.com/IntelRealSense/librealsense/issues/2850

MartyG-RealSense, 2019-09-07:
A bug-fix for Kernel 4.19 was added in SDK version 2.24.0.  Some users continued to report problems with using 4.19 though.

A RealSense user succeeded in using 4.19 with Pi4 and Raspbian Buster with a method suggested by a member of Intel's RealSense team.

https://github.com/IntelRealSense/librealsense/issues/4565#issuecomment-518359584

dorodnic, 2019-09-08:
We only support official kernel versions used in i86 LTS versions of Ubuntu 16 / 18. This doesn't mean it will or will not work on non-i86 hardware, we just cannot make such commitments. For non-i86 platforms we recommend [libuvc_installation.md](https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md) installation without kernel patching.

coriolan-cs, 2019-09-13:
I've got it working as well on a Pi4 with Raspbian Buster 4.19

AnupamYedida, 2019-10-05:
Is the IMU data working as well ?

coriolan-cs, 2019-10-07:
I'm using the D410 module, it doesn't have an IMU, so I can't say unfortunately.

rofreli, 2019-10-07:
I'm using Ubuntu 18 in my Pi 4. With some adjustments, realsense works but Ubuntu does not have official kernel to Pi 4 then some drivers like WiFi need some attention.

MartyG-RealSense, 2019-10-08:
Apparently the STM32MP1 uses ARM Cortex-A7 for its compatibility with open-source operating systems such as Linux and Android.  There was a case where somebody was successfully using T265 with a Cortex-A7 processor and Ubuntu. 

https://github.com/IntelRealSense/librealsense/issues/4408#issuecomment-525725002

 It looks as though the BananaPi Zero hardware they were using was running Kernel 4.19.0.  

http://wiki.banana-pi.org/Banana_Pi_BPI-M2_ZERO

[image]

alexcherpi, 2019-10-10:
Thanks MartyG, 
This maybe a stupid question, but is it possible to dialog to the Realsense T265 with a MCU, e.g STM32F7 which has a USB 2.0 OTG ? Which steps would be required ?

MartyG-RealSense, 2019-10-10:
The T265 can stream pose data on USB2 but needs USB3 for fisheye streams.  So USB2 OTG would not be appropriate if you need fisheye data in your project.

I don't know about connecting T265 to OTG, but it is certainly possible to connect RealSense depth cameras to OTG if you use an OTG adapter to connect a full-size USB A connector to an OTG micro-port.

alexcherpi, 2019-10-10:
So if we only need the pose data you mean that we could get it using the USB 2.0 port of the STM32F7 for example? But is there any library to know how to initiate this communication and to parse the received data?

alexcherpi, 2019-10-10:
Thanks for your help.
The STM32F7 has only one Arm® Cortex®-M7 core inside. Do you have a link to the API you mentionned?

MartyG-RealSense, 2019-10-10:
If you scroll right to the bottom of the page and then scroll a little up, the API section is near the bottom of the page.

RealSenseCustomerSupport, 2019-10-31:
----------------------------------------------

Hi All

Is there more that is needed on this thread?  Thanks to @MartyG-RealSense for links to other threads with updates on this topic.

closed 2019-11-19 by RealSenseCustomerSupport