realsenseai/librealsense #4819: how to compress depth frame data

tom8zds, 2019-09-07:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |SR300  |                                      | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C}                          | 

### Issue Description
[image]
I'm working on a project which I need to stream depth data through Wi-fi, but it seems no  bandwidth can support it.
the .ply files saved have a significant volume difference. The viewer export 600KB while export_to_ply give out 2MB. If it is possible to compress the depth data to 600KB per frame, 50Mb bandwidth is enough for streaming

MartyG-RealSense, 2019-09-07:
You are correct that there are challenges to streaming over wi-fi.  Intel's RealSense development team are actively working on methods to improve compression.  

Wi-Fi transmission has been accomplished in the past by using drone communication methods such as GStreamer.
https://software.intel.com/en-us/articles/intel-aero-compute-board-and-intel-realsense-technology-for-wi-fi-streaming-of-rgb-data

Otherwise, it may be more practical to send the data over  wired Ethernet connection, which RealSense has significant support for these days.

https://dev.intelrealsense.com/docs/depth-camera-over-ethernet-whitepaper

RealSenseCustomerSupport, 2019-10-09:
----------------------------------------------

Hi @tom8zds,

Any update? It's much reliable to transfer the data over the ethernet. The traffic in wifi environment is full of interference in today's working environment.
Since the bandwidth and interference is the concern, lowering depth resolution and FPS is another option.

RealSenseCustomerSupport, 2020-04-22:
----------------------------------------------

Hi.

Support for RS2_FORMAT_Z16H has been added as of Version 2.32.1 release. Will you be needing further help with this? If we don’t hear from you in 7 days, this issue will be closed.

Thanks

closed 2020-04-29 by RealSenseCustomerSupport