realsenseai/librealsense #4820: Eporting to PLY in Unity

visus100, 2019-09-07:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | FW_5_11_11_100 | 
| Operating System & Version |   Win 10 | 
| Platform                                 | PC |
| SDK Version                            | SDK-WIN10-2.23.0.909                     | 
| Language                            |  C#/unity                        | 

### Exporting to PLY in unity

I'm trying to figure out how to export through editing script RsPointCloudRender.cs

### Code:

private int counterHelperInt = 0;

protected void LateUpdate() {
        if (q != null){
            Points points;
            
            if (q.PollForFrame<Points>(out points))  using (points) {
                    
                    if (points.Count != mesh.vertexCount)  {
                        using (var p = points.GetProfile<VideoStreamProfile>()) {
                            ResetMesh(p.Width, p.Height);
                        }
                    }

                    if (points.TextureData != IntPtr.Zero) {
                        uvmap.LoadRawTextureData(points.TextureData, points.Count * sizeof(float) * 2);
                        uvmap.Apply();
                    }

                    if (points.VertexData != IntPtr.Zero) {
                        points.CopyVertices(vertices);
                        
                        mesh.vertices = vertices;
                        mesh.UploadMeshData(false);

                        if (counterHelperInt <= 101) {
                            counterHelperInt++;
                        }

                        if (counterHelperInt > 10 && counterHelperInt < 100) {
                            Debug.Log(counterHelperInt);

                            VideoFrame frame2;
                            if (q.PollForFrame<VideoFrame>(out frame2)) using (frame2) {
                                 Debug.Log("Saved mesh ");
                                 points.ExportToPLY("filename.ply", frame2);
                            }
                        }
                    }
                }
        }
    }

### Error
I am getting this error:

`
ExternalException: rs2_export_to_ply(frame:000000008AC1B140, fname:f)
Rethrow as ArgumentException: frame must be video frame
Intel.RealSense.ErrorMarshaler.MarshalNativeToManaged (System.IntPtr pNativeData) (at <441657b0dbbd43269996475b2c92e0df>:0)
Intel.RealSense.Points.ExportToPLY (System.String fname, Intel.RealSense.Frame texture) (at <441657b0dbbd43269996475b2c92e0df>:0)
RsPointCloudRenderer.LateUpdate () (at Assets/RealSenseSDK2.0/Scripts/RsPointCloudRenderer.cs:201)
`

I couldn't find any example in Unity which could solve this problem. Is this bug or i'm doing something wrong? Could anyone help?

wodndb, 2019-09-30:
This issue maybe help you. https://github.com/keijiro/Pcx/issues/23

RealSenseCustomerSupport, 2019-10-21:
----------------------------------------------

Hi @visus100,

Has the link provided by @wodndb helped to find an answer to your question?

Thank you!

RealSenseCustomerSupport, 2019-10-25:
----------------------------------------------

@visus100,

Do you still need help with this question?

closed 2019-10-31 by RealSenseCustomerSupport

Max-Stricker, 2021-07-08:
I have the same problem

wodndb, 2021-07-10:
https://github.com/wodndb/ExportPlyForRealSense

I create example project for export point cloud to ply.

UnityDevSan, 2021-07-12:
Wow, Thank you very much that's awesome! 
But i get it right, it's just possible to capture a single frame and not a whole scanning session? (like walk around a object)

wodndb, 2021-07-12:
@UnityDevSan 
Oh.. That's not support. In this issue just for converting Realsense point cloud to ply format. You should find ICP algorithm and SLAM for scanning something.

wodndb, 2021-07-13:
@UnityDevSan Do you want to convert streamed frames to ply files? I'll try to test in this week. But I can't sure to convert ply file in every frames.

UnityDevSan, 2021-07-13:
Would be awesome, i will try some other github projects too

UnityDevSan, 2021-07-21:
Hey, i couldn't find something similar with the RealSense Unity SDK, did you find something out?