realsenseai/librealsense #4821: RealSense D435i - unusable depth frames after using the device outside

theodop, 2019-09-08:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {D435i} | 
| Firmware Version                   | 05.11.06.250 | 
| Operating System & Version |   Ubuntu 18  | 
| Kernel Version (Linux Only)    |  4.18.0-25-generic                                        | 
| Platform                                 | PC  |
| SDK Version                            |  2, RealSense Viewer                          | 
| Language                            |  C++ / RealSense Viewer                    | 
| Segment			| Depth                       | 

Following a firmware update to version 05.11.06.250, I am experiencing severe issues with the depth frames after using the device outdoors.

To reproduce, I use the device either through my own code, or through the RealSense viewer. I go outside, and then I go back inside. After this process, every depth frame returns has almost no density until the pipeline is restarted. The images below show before and after going outside.

I am using the default auto-exposure settings in the RealSense viewer. In code, I am not altering the device configuration at all after opening the pipeline.

It may be related to the emitter, as after going outside and coming back inside I can no longer see IR being emitted if I look at the detector.

[image]
[image]

MartyG-RealSense, 2019-09-08:
There have been a couple of past cases I know of where the depth has been badly affected by light when outdoors (e.g freezing or becoming very sparse) and a reset of the pipeline corrected the problem.

https://support.intelrealsense.com/hc/en-us/community/posts/360033445713-D435-Sunlight-saturation-problem

Some advice by a member of Intel's RealSense team (agrunnet) is given here:

https://github.com/IntelRealSense/librealsense/issues/2875#issuecomment-446788177

https://github.com/IntelRealSense/librealsense/issues/4022

Also, a new firmware, 5.11.11.100,  was released a couple of weeks ago and is now the current recommended firmware on the SDK Releases download page, so updating to that firmware may be worth a try.

theodop, 2019-09-08:
Thank you Marty, I agree that it appears to be the same issue as #4022. I only checked the new production build; I haven't had a look at the most recent dev build. I will do this and update.

theodop, 2019-09-15:
The later firmware version resolved the issue.

closed 2019-09-15 by theodop