realsenseai/librealsense #4822: SR305 Time Domain

tonyhsiao1995, 2019-09-09:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.11.06.250 | 
| Operating System & Version |   Ubuntu16.04  | 
| Kernel Version (Linux Only)    | 4.15.0-55-generic  | 
| Platform                                 | PC |
| SDK Version                            |    2.23.00   | 
| Language                            |  C++  | 

### Issue Description
In the Realsense viewer, the D415 and D435 timestamp are global time clock. However, SR305 is using the hardware time clock. How do I fix it?

MartyG-RealSense, 2019-09-09:
The release notes for SDK 2.23.0, where global time was added, imply that global time support is dependent on sensor support for it.  I checked the RealSense Viewer with both a D435i and an SR300, and the Viewer offered a "Global Time Enabled" box-toggle option at the bottom of the **Controls** options category for the D435i to toggle global time on-off but did not offer that option on the SR300's version of the options layout.

More information on global time can be found here:

https://github.com/IntelRealSense/realsense-ros/issues/796

RealSenseCustomerSupport, 2020-03-18:
----------------------------------------------

Hi @tonyhsiao1995 Please note that global timestamp was added for the D400 feature of multi-camera. It doesn't apply to SR300. RGB-Depth is supported by the HW TS.
If we don't hear from you in 7 days, we will close this issue.

closed 2020-03-27 by RealSenseCustomerSupport