realsenseai/librealsense #4824: librealsense building examples problem

LJS98, 2019-09-09:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | RealSense Viewer 2.28 | 
| Operating System & Version |  Linux (Ubuntu 18.04 LTS)   | 
| Kernel Version (Linux Only)    | 5.0.0.27.84            | 
| Platform                                 | PC |
| SDK Version                            | 2.0         | 
| Language                            |  c++                 | 
| Segment			|  building examples              | 

### Issue Description
Hi, I'm starting a project with realsense d435. So I tried to build examples. I followed installation.md(linux). Also I did cmake whole sdk (librealsense-master/build$ cmake .. && make).
After then, I did cmake pointcloud in examples. I tried make command but there is nothing happen. I want a executable file.
[image]

I think "if(BUILD_GRAPHICAL_EXAMPLES)" is the problem which is in CMakeLists.txt. maybe...
(if I command           glxinfo | grep "OpenGL version"
ouput is           OpenGL version string: 4.3 (Compatibility Profile) Mesa 19.0.8)
How can i solve this problem?

I would appreciate any help.
Thank you!

MartyG-RealSense, 2019-09-09:
The link below contains details of the customization flags for CMake, such as BUILD_GRAPHICAL_EXAMPLES.

https://github.com/IntelRealSense/librealsense/wiki/Build-Configuration

That particular command, which is set to True by default, excludes the building of tools that are dependent on OpenGL when set to False, and is suitable for use with 'headless' systems.   Your glxinfo output would suggest to me that you have OpenGL 4.3 installed though.

did you use the build flag **-DBUILD_EXAMPLES=true** when building the SDK, please?

LJS98, 2019-09-09:
Thank you for helping me. However, I can't still get an exe file when I make.

ev-mp, 2019-09-09:
@LJS98 , building all SDK artifacts, including the library, examples, wrappers, shall be invoked from the root librealsense directory as defined in the [installation guide](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md#building-librealsense2-sdk)
> Navigate to librealsense root directory and run mkdir build && cd build....

Please follow the guide's instructions and run the build with the examples flag as suggested by @MartyG-RealSense:
 `cmake .. -DBUILD_EXAMPLES=true  -D... && make -j && sudo make install`

LJS98, 2019-09-09:
@ev-mp 
I tried it but it is same.
There is not error when I command 'make' but also there is not any exe file.

ev-mp, 2019-09-09:
@LJS98 , it's not clear what is the current state of your workspace after all the tries.
I'd suggest to make a clean rebuild and provide the full console log for review if you'd still encounter the issue. Should be equivalent to

Navigating to librealsense root dir
>cd build
sudo make uninstall && make clean && sudo rm -rf * && cmake .. -DBUILD_EXAMPLES=true -D<put_all_the_requested_flags> && make -j$(nproc) && sudo make install

LJS98, 2019-09-10:
This is my console log
As you can see, rs-hello-realsense works. But pointcloud stiil doesn't work.
[console_log.txt](https://github.com/IntelRealSense/librealsense/files/3593110/console_log.txt)

closed 2019-09-10 by LJS98