realsenseai/librealsense #4827: streaming color and infrared from two different sources

mstranne, 2019-09-09:
Hello,

I am using the R200 Camera on windows with librealsense in an cpp library for tracking of an infrared light pattern, which is working fine. I am able to use the provided extrinsics  to convert the pose from IR frame to Color, all working great!

I am wrapping all stuff to use it inside Unity, which was also easy to pass the color stream up to C# level.

However, now I want to use another library for marker tracking inside the Color stream. When passing camera id 1 the lib is able to recognize the Color Camera via USB and it works perfectly like with the built-in laptop camera.

Nevertheless, as soon as I am now using both libraries for tracking, the secound one is not able to start a camera stream.

Am I able to use the Color Camera and the Infrared camera in several applications?
And how would I do that?

ev-mp, 2019-09-18:
@mstranne hello, unfortunately this use-case was not implemented in Librealsense**1** where ony one executable had control over all the connected depth cameras.
So when you start streaming the depth sensor all the other (including RGB) are not available to the other executables.

You may work/around by using a "master" application to stream depth+ir+color and then pass the RGB frame to a 3rd-party  using shared memory or a similar method.

RealSenseCustomerSupport, 2019-10-09:
----------------------------------------------

Hi @mstranne, please let us know if further help is needed.
Thank you.

closed 2019-10-17 by RealSenseCustomerSupport