realsenseai/librealsense #4828: Getting Depth Data in Unity to map to heightmap of the Terrain in Unity.

anuragtiwari, 2019-09-09:
Camera: D415
Unity 2018.1.6f1 
C#

I am trying to generate a runtime terrain from the depth values of the realsense camera D415. 
after going through multiple related issues posted, I went through the depthCutoff.cs script but do not understand where(which scen eit is getting used to map depth values to something

Here is what i have gotten so far. 
I have created a terrain which is 640x480, the height of which will come from the depth sensor - i need a float array of 640x480.. I followed the c# cookbook and some forum links and issues here and gott he massive 1D depth array. 

but the error index out of range is obvious here since the 1D depth array is 2764800 items longs which is 640*480(resolution of the depth frame) * 9 (what is this 9) ? how do i map it to 640x480. It would really help if there is actual Unity based guidance for the unity wrapper. the getting started just shows gameobjects with a maze like pipeline flow. 
how do i get 640time480 float values from 2764800 values. 

`using UnityEngine;
using Intel.RealSense;

/// <summary>
/// this script is made to test the new terrain in unity 
/// </summary>
public class TerrainGenerator : MonoBehaviour {

    public int depth = 40;
    public int width = 640 ;
    public int height =480;

    public float scale = 20f;

    public float xOffset = 10;
    public float yOffset = 10;

    
    public bool scroll;
    public float scrollSpeed;

    private float[] vertices;
    private Pipeline pipe;
    private PointCloud pc;

    private void Start()
    {
        //realsense 1d array
        pipe = new Pipeline();
        pc = new PointCloud();

        pipe.Start();

        using (var frames = pipe.WaitForFrames())
        using (var depth = frames.DepthFrame)
        using (var points = pc.Process(depth).As<Points>())
        {
            // CopyVertices is extensible, any of these will do:
            vertices = new float[points.Count * 3];
            // var vertices = new Intel.RealSense.Math.Vertex[points.Count];
            // var vertices = new UnityEngine.Vector3[points.Count];
            // var vertices = new System.Numerics.Vector3[points.Count]; // SIMD
            // var vertices = new GlmSharp.vec3[points.Count];
            //  var vertices = new byte[points.Count * 3 * sizeof(float)];
            points.CopyVertices(vertices);

        }
        Debug.Log(vertices.Length + " : is the length of the array");

        xOffset = Random.Range(0f, 9999f);
        yOffset = Random.Range(0f, 9999f);

        scroll = false;
        scrollSpeed = 0;
        

        Terrain terrain = GetComponent<Terrain>();
        terrain.terrainData = GenerateTerrain(terrain.terrainData);

        
    }
    void LateUpdate()
    {
       
        if (scroll)
            xOffset += Time.deltaTime * scrollSpeed;
        
    }
    TerrainData GenerateTerrain(TerrainData tData)
    {
        tData.heightmapResolution = width + 1;
        tData.size = new Vector3(width, depth, height);
        //tData.SetHeights(0, 0, GenerateHeights());
        tData.SetHeights(0, 0, Convert1DTo2DArray());
        return tData;
    }
    float[,] GenerateHeights()
    {
        float[,] heights = new float[width, height];
        for (int x = 0; x < width; x++)
        {
            for (int y = 0; y < height; y++)
            {
                heights[x, y] = CalculateHeight(x, y);
                //later will become the depth value from the real sense camera
                
            }
        }
        return heights;
    }

    float CalculateHeight(int x, int y)
    {
        
        float xCoord = (float)x / width * scale + xOffset;
        float yCoord = (float)y / height * scale + yOffset;
        
        return Mathf.PerlinNoise(xCoord, yCoord);
       
        
    }

    float[,] Convert1DTo2DArray()
    {
        int width = 640;
        int height = 480;
        var Map = new float[width, height];
        for (int i = 0; i < width; i++)
        {
            for (int j = 0; j < height; j++)
            {
                Map[j, i] = vertices[j * 10 + i];
            }
        }
        return Map;
    }
    

}
`

RealSenseCustomerSupport, 2019-10-05:
----------------------------------------------

Hi @anuragtiwari,

Can you please explain multiplication by 3 at this line of the above-mentioned code :
vertices = new float[points.Count * 3];

Thank you!

anuragtiwari, 2019-10-06:
It’s part of the code provided by the C# cookbook for real sense under generic support  and was my attempt at getting the vertices. 
If you remove all the lines above offset in the start function then it will still work and generate the terrain with perlin noise. 
My goal is to get vertices from the depth sensor into a usable vertex array 
since the point cloud scene in unity does not form triangles at all in the mesh. 
Is there any way to do that if not the above? I would not mind if it’s a mesh instead of a terrain because all I need is a usable vertex height array which makes sense

RealSenseCustomerSupport, 2019-10-08:
----------------------------------------------

@anuragtiwari,

Please try obtain a depth array of the right size by removing multiplication by 3 at this line:
vertices = new float[points.Count * 3];
After you will get it, you'll be able to map the items to the resolution you are interested in.

Thank you!

anuragtiwari, 2019-10-10:
same errors along with an extra index out of range at pc.Process(depth)

anuragtiwari, 2019-10-11:
im going to close this issue since i have moved onto a difference library and sensor. its is just sad that the d400 sensors are so much better for extended depth but this library feels rushed and incomplete. (there are comments in the scripts and shaders that have "TODO" in them.). 
while my incompetence is a big factor in me having to choose a different sensor, if i have to implement marching cubes and quickhull to convert Meshtopology from Points to Mesh without a single line of comment in the code, that even the Realsense support doesnt know, then that would be a library I would not release to the public or publicize with Unity being supported. 
Please bring back the tutorials and actual guides as to how to better use the library. the point cloud generation is highly incomplete specially with the filters. It makes me so sad that i have to use a inferior sensor for my purpose because the library for it is more usable (and that is saying something since the codebase is from OpenCV).

closed 2019-10-11 by anuragtiwari

ogoshen, 2019-10-15:
Hi @anuragtiwari, and sorry for the late response.

The problem with the code above seems to be just a case of miss-configuration and not interpreting the data properly.

Unless configured otherwise, the default depth resolution is 720p, not 480p (which was hardcoded here, and could instead be queried).
You should be able to assert this yourself by inspecting the `RsDevice` component in the editor (in play mode) .

The point cloud code from [the cookbook](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/csharp/Documentation/cookbook.md#generics-support) is correct, and will output vertices as float triplets, so 1280&times;720&times;3=2764800.

@RealSenseCustomerSupport threw you off the trail here, but that's an easy mistake to make...

Otherwise, can't see anything wrong with the code.

I'll now address the use-case problem, seems like what you're interested in is visualizing the point  cloud as a mesh. I really doubt that Unity's terrain renderer is what you are looking for, especially if you plan on updating it per frame. Terrains are best suited for static high resolution data.

Now, it's also very possible to generate dynamic meshes at runtime, this could even be easier and more efficient in Unity [soon](https://forum.unity.com/threads/feedback-wanted-mesh-scripting-api-improvements.684670/). And there are other tricks you could do today, like pre-generating the mesh and displacing the vertices in a shader using Vertex Texture Fetch (`tex2dlod`), much like how simple heightmap rendering can be done. 

There are some caveats here though:
* terrains usually don't have holes, which depthmaps have.
* terrains are usually fairly continuous\smooth, which depthmaps aren't - two neighboring pixels could belong the different objects and have (or be missing) very different depths.
 
So the usual heightmap visualization isn't really suited here, it will cause rendering artifacts 

These issues, as well as the current performance overhead in creating meshes in Unity (regular dynamic meshes, that can have proper mesh colliders for instance) are why I usually prefer not to visualize depthmaps as meshes, and why I haven't supplied such a component in the wrapper.

I'll also mention that the supplied point cloud shader has a few configurable parameters, mainly the point size, and the [scale](https://github.com/IntelRealSense/librealsense/blob/81d469db173dd682d3bada9bd7c7570db0f7cf76/wrappers/unity/Assets/RealSenseSDK2.0/Shaders/PointCloudGeom.shader#L7) toggle, which will cause points to be distance scaled and behave more like a mesh 

AFAIK visualizations based on meshes from marching cubes are backed by volumetric data, doing pose-estimation, volume integration and mesh reconstruction is currently out of scope for this SDK.

For more point  cloud visualizations you might be interested in @keijiro's fantastic experiments like https://github.com/keijiro/Rsvfx.

I'm sure the docs could improve, but note that the .NET & Unity wrappers aren't rehashing everything about the SDK, and focus mostly on their own specifics.  

Hope this helps.

anuragtiwari, 2019-10-21:
Hi Ogoshen, 
Firstly thank you! 
What started as a fun summer project became a nightmare to achieve. So thank you for taking the time to look into my issue. i really really appreciate it. 
→ I finally figured i needed to use a mesh(terrain so far has gotten me the furthest - more below) so i can pass the vertices directly. But that didnt work either. (attaching image - the right side is when i pass the entire mesh to another but that still is a point cloud, the left image is when i pass vertices). - which brings me to my next point
[image]
→ RS Device is set to 640 * 480. But my vertex array length is 19200 which is 160* 120. And this is from the scene Point Clouds processing block code where i just added a debug.log

 > And there are other tricks you could do today, like pre-generating the mesh and displacing the vertices in a shader using Vertex Texture Fetch (tex2dlod), much like how simple heightmap rendering can be done.

→ Again Im not sure what I am doing wrong but the vertices hate me.  any resources i can look at for ↑
→ Yes I looked at the geometry shader that does indeed have the point size scale which would allow me to blow up the point cloud but if i try to pass any texture in that shader to a point in the cloud it destroys the texture and looks like minecraft pixel art (not always a bad thing, but not suited for my purposes).
→ I honestly wouldn't mind the overhead and completely understand why it would be a bad idea to do it specially without threading.But, If there is a way I would love to know how since I can write a time step for the update or thread it and would be happy to share it here. Just need a mesh with triangles that I can write a shader for and pass textures to. 
→ If I can get even the Vector3 position one vertex properly in the scene, i'd be happy. 
→ I did pass the values to a terrain and am attaching code and image for how that looks too comes out as a triangle. Again the terrain is set to 640 * 480 but does this weird slanted triangle thing where the Texture doesnt allign with height and the terrain stays the way bigger. 
[image]

### → Code for Passing vertices to terrain 

`
[RequireComponent(typeof(MeshFilter), typeof(MeshRenderer))]
public class RsPointCloudRenderer1 : MonoBehaviour
{
    public RsFrameProvider Source;
    private Mesh mesh;
    private Texture2D uvmap;

    [NonSerialized]
    private Vector3[] vertices;

    FrameQueue q;

    float[] heights;
    
    float[,] heights2D;

    public int t_depth = 200; //height on the y axis
    public int t_width = 640; // x axis -> width
    public int t_height = 480; //this is the length of our terrain - > more line Z axis
    public GameObject TerrainToPlayWith;
    void Start()
    {
        Source.OnStart += OnStartStreaming;
        Source.OnStop += Dispose;
        heights2D = new float[640, 480];
    }
    //rest of the methods are same

    protected void LateUpdate()
    {
      ...

                    if (points.VertexData != IntPtr.Zero)
                    {
                        points.CopyVertices(vertices);

                        mesh.vertices = vertices;
                        heights = new float[mesh.vertices.Length];
                        // Debug.Log("We are here" + heights.Length);
                         for(int i =0;i<vertices.Length;i++)
                         {
                              heights[i] = vertices[i].y;
                              
                         }
                        for (int i = 0; i < t_width; i++)
                            for (int j = 0; j < t_height; j++)
                                heights2D[i,j] = heights[(j * t_width) + i];
                        Terrain terrain = TerrainToPlayWith.GetComponent<Terrain>();
                        //terrain.terrainData = GenerateTerrain(terrain.terrainData);
                        terrain.terrainData.heightmapResolution = t_width + 1;
                        terrain.terrainData.size =  new Vector3(t_height, t_depth, t_width);
[... 212 lines cut ...]
						return DoBlending(7, TextureFloat, e1Color, e2Color);
					}
					if (TextureFloat < 9)
					{
						fixed4 e2Color = tex2D(_e2, i.texcoord);
						fixed4 e3Color = tex2D(_e3, i.texcoord);

						return DoBlending(8, TextureFloat, e2Color, e3Color);
					}

					fixed4 e3Color = tex2D(_e3, i.texcoord);
					
					return e3Color;

					fixed4 DeepWaterColor = tex2D(_DeepWater, i.texcoord);
					fixed4 ShallowWaterColor = tex2D(_ShallowWater, i.texcoord);

					return lerp(DeepWaterColor, ShallowWaterColor, i.blend.w);

				}
			ENDCG
			}
		}
	FallBack "Diffuse"
}`