realsenseai/librealsense #4832: Are there any sample code for Intel® RealSense™ cameras with python like this page?

Dontla, 2019-09-10:
Like the sample code in this page: [https://github.com/IntelRealSense/librealsense/tree/development/examples#list-of-examples](https://github.com/IntelRealSense/librealsense/tree/development/examples#list-of-examples), I prefer to use c/c++ language, but I can't understand them...
[image]

MartyG-RealSense, 2019-09-10:
A list of example Python programs can be found on the link below.

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python/examples

There are also the Jupyter Notebook example programs.

https://github.com/IntelRealSense/librealsense/tree/jupyter/notebooks

Dontla, 2019-09-10:
@MartyG-RealSense Yeah, does Intel Realsense SDK have the Python interfaces? I was confused...

Dontla, 2019-09-10:
I have seen some .pyd file in the Intel Realsense folders, but I don't know how can I use these critical files...can you tell me how to do...

MartyG-RealSense, 2019-09-10:
You do not need to be concerned about accessing those files.  Python support in the RealSense SDK is provided by the 'Pyrealsense2' Python wrapper, which has to be set up.  The easiest way to set up Pyrealsense2 is with the Pypi instruction **pip install pyrealsense2**

Full installation instructions can be found here:

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python

dorodnic, 2019-09-10:
Here is the list of python example - [python/examples#list-of-examples](https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python/examples#list-of-examples)

RealSenseCustomerSupport, 2019-10-01:
----------------------------------------------

Hi @Dontla
Do you need further assistance on this topic?

Dontla, 2019-10-02:
Thank you! I dt need further more help for the time being.

closed 2019-10-04 by RealSenseCustomerSupport

Najmuleee, 2020-11-18:
when i import pyrealsense2 as rs it shows no module. could you tell me how to install pyrealsense2 in jupyter notebook.

MartyG-RealSense, 2020-11-18:
Hi @Najmuleee   If you are using Jupyter Notebook and experiencing this error:

**DLL load failed while importing pyrealsense2: The specified module could not be found.**

then the comment in the link below may be helpful.

https://support.intelrealsense.com/hc/en-us/community/posts/360049911373/comments/360012797974