realsenseai/librealsense #4833: build error ubuntu 18.04 on jetson nano

mickeyouyou, 2019-09-10:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
build error 

-- Internet connection identified
-- Info: REALSENSE_VERSION_STRING=2.24.0
-- Setting Unix configurations
-- using RS2_USE_V4L2_BACKEND
-- pybind11 v2.2.1
-- Building with TM2
CMake Error at third-party/libtm/fw/CMakeLists.txt:22 (file):
  file DOWNLOAD HASH mismatch

    for file: [/home/geek-car/librealsense/build/third-party/libtm/fw/target-0.0.18.5715.mvcmd]
      expected hash: [cc12cf05f387d80f65e98c4d41fc883725124a08]
        actual hash: [58a8e4db14a5941da34f8258fe20ded27b02693c]
             status: [56;"Failure when receiving data from the peer"]

Call Stack (most recent call first):
  third-party/libtm/fw/CMakeLists.txt:47 (target_binary)

-- Download TM2 firmware 0;"returning early; file already exists with expected SHA1 hash" for central_app-2.0.19.271.bin
-- Download TM2 firmware 0;"returning early; file already exists with expected SHA1 hash" for central_bl-1.0.1.112.bin
-- Fetching recommended firmwares:
-- D4XX_FW_VERSION: 5.11.6.250
-- SR3XX_FW_VERSION: 3.27.3.0
-- Download firmware 0;"returning early; file already exists with expected SHA1 hash" for D4XX_FW_Image-5.11.6.250.bin
-- Download firmware 0;"returning early; file already exists with expected SHA1 hash" for SR3XX_FW_Image-3.27.3.0.bin
-- Configuring incomplete, errors occurred!
See also "/home/geek-car/librealsense/build/CMakeFiles/CMakeOutput.log".
See also "/home/geek-car/librealsense/build/CMakeFiles/CMakeError.log".

ev-mp, 2019-09-10:
@mickeyouyou , the error mean that a required resource file was not downloaded correctly ( hashes don't match).
Most probably there was a temporal glitch in the firmware hosting server communication or a local network issue, thus rerunning CMake should probably fix it.

Also make sure that the specified file is not locked by OS, and delete it manually if it is before running CMake
>    for file: [/home/geek-car/librealsense/build/third-party/libtm/fw/target-0.0.18.5715.mvcmd]

mickeyouyou, 2019-09-10:
thank you so much. I found it after I have been written this issue.

closed 2019-09-10 by mickeyouyou

beyond21299, 2019-12-27:
> thank you so much. I found it after I have been written this issue.

How do you handle this cmake error？