realsenseai/librealsense #4834: Timestamp Domain

Dontla, 2019-09-10:
Hi, I have updated my Intel Realsense SDK vertion to v2.25.0, but when I opened my camera in Intel Realsense viewer, it comes to this red warning :
[image]
[image]
Is this warning a critical warning? how can I solve it?

dorodnic, 2019-09-10:
Hi @Dontla 
This is not a critical warning (it has no effect on depth quality) However, having more accurate timestamps can be beneficial in some applications. Here is the relevant documentation for Windows - [installation_windows.md#use-automation-script](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_windows.md#use-automation-script)

RealSenseCustomerSupport, 2019-10-01:
----------------------------------------------

@Dontla, please kindly let us know if further help is needed for this ticket. Thank you.

Dontla, 2019-10-02:
Thank you! The problem has been solved.

closed 2019-10-02 by RealSenseCustomerSupport