realsenseai/librealsense #4835: RuntimeError: Frame didn't arrived within 5000

Dontla, 2019-09-10:
when I ran the quick start demo in this [page](https://pypi.org/project/pyrealsense2/#description), it came to this error, what problems cause this error possible? QAQ
[image]

kjacks21, 2019-09-10:
You haven't specified a data source. For example, if you want to read in a .bag file, do the following to start:

```
config = rs.config()
rs.config.enable_device_from_file(config, bag_fname, repeat_playback=False)
pipeline = rs.pipeline()
config.enable_stream(rs.stream.color)
pipeline.start(config)
```

RealSenseCustomerSupport, 2019-10-02:
----------------------------------------------

Hi @Dontla,

Can you please provide us with the details of your environment configuration:
Operating System & Version
Kernel Version (Linux Only)
Platform
SDK Version
The source of frames (camera, .bag file)
If the source is a camera: how many cameras are in active use in the system?
For each camera please provide following information: Camera Model,Firmware Version.

Thank you!

RealSenseCustomerSupport, 2019-10-22:
----------------------------------------------

Hi @Dontla,

Do you still need help with this question?

RealSenseCustomerSupport, 2019-10-28:
----------------------------------------------

@Dontla,

The example you have issue with should work without any hassle. Please let us know if you need any further assistance with it and if you do need assistance, please provide us with the information about your environment configuration.

closed 2019-11-04 by RealSenseCustomerSupport