realsenseai/librealsense #4837: What is the C# equivalent of the C++ align.process() class call?

brettmilinski, 2019-09-10:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.11.06.250 | 
| Operating System & Version |   Win (10)  |  
| Platform                                 | PC|
| SDK Version                            |  2.25.0                         | 
| Language                            |  C#                          | 

### Issue Description
I am creating a C# version of the rs-dnn example solution provided by Intel RealSense. 
In rs-dnn, there are a few lines that align the frames of the color stream:

`rs2::align align_to(RS2_STREAM_COLOR); `
`// Wait for the next set of frames`
`auto data = pipe.wait_for_frames();`
`// Make sure the frames are spatially aligned`
`data = align_to.process(data);`

I was wondering, what is the C# equivalent to this? Do I have to process each frame in a custom processing block or is there an easier way to apply the alignment to my frames?

MartyG-RealSense, 2019-09-10:
I hope that the link below will be of help to you.

https://github.com/IntelRealSense/librealsense/issues/4719

brettmilinski, 2019-09-10:
@MartyG-RealSense That was what I needed. Thanks for the help, I appreciate it!

closed 2019-09-10 by brettmilinski