realsenseai/librealsense #4838: Pyrealsense2 ROI

BenDavisson, 2019-09-10:
Camera Model                       | D435 | 
Firmware Version                   | 0.5.11.06.250 | 
Operating System & Version |  Windows 10  | 
Platform                                 | PC |
SDK Version                           |  2.25.0 | 
Language                               | python | 
Segment			               |  Machine vision   | 

Is there a way to define an ROI using Pyrealsense2 with my camera model?  

I know it is possible using OpenCV, however, my issue is a little more advanced.  The problem is that the stereo cam is determine depth based off of objects that aren't important.  Meaning, the closer objects to the cam that are on the far edges of the view are changing the color of the stereo output in an undesirable way.  

Is there a way to set an ROI so the only depth detection that takes place is the in the camera field of view?

Thanks!

MartyG-RealSense, 2019-09-10:
I hope the link below will be of help regarding setting a ROI with Python.

https://github.com/IntelRealSense/librealsense/issues/2681#issuecomment-436391721

BenDavisson, 2019-09-10:
Sorry if I misunderstand, but doesn't the linked issue only handle setting an autoexposure ROI?

I'm trying to configure an ROI for the entire camera.  For example, [OpenCV can return a region of interest](https://docs.opencv.org/master/d3/df2/tutorial_py_basic_ops.html)  of an image with the correct gird points(click link, then scroll down or search for ***Image ROI*** in link).

Is this possible to do using Pyrealsense2? In other words, configure the camera to only return the view of a region of interest?

MartyG-RealSense, 2019-09-10:
You did not mention if you are using a point cloud.  If you are, the discussion below on generating a point cloud with Python and cropping it to the ROI may help.

https://github.com/IntelRealSense/librealsense/issues/2769

BenDavisson, 2019-09-11:
Thank your for the timely responses Marty! I really appreciate it.

To my understanding... I am not using a point cloud.  I am simply trying to define a region of interest for the camera to use, just like you can do with other machine vision camera SDK's.

MartyG-RealSense, 2019-09-11:
In RealSense SDK 2.0, the term 'region of interest' is typically associated with the auto-exposure.  Another kind of ROI is to generate a bounding-box and then occlude the data outside of that bounding box.     Here is the approach that one RealSense user took to implementing this:

https://github.com/IntelRealSense/librealsense/issues/2016#issuecomment-403804157

RealSenseCustomerSupport, 2019-11-01:
----------------------------------------------

Hi @BenDavisson,

Is the issue resolved per @MartyG-RealSense's comment? So far, we don't provide this kind of feature to specify a defined depth region.

closed 2019-12-05 by RealSenseCustomerSupport