realsenseai/librealsense #4840: Writing .bag stream to video gives inconsistent frame counts

kjacks21, 2019-09-10:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | latest (no info tab in viewer) | 
| Operating System & Version |   Ubuntu 18  | 
| Kernel Version (Linux Only)    |  5.0.0-27-generic                                      | 
| Platform                                 | PC  |
| SDK Version                            |  2                         | 
| Language                            |  python                          | 
| Segment			|                        | 

### Issue Description
Using the python wrapper, when I write the bag file to a video file (e.g., mp4), the resulting video varies in the total frame count by a few frames. For example, the video from the same bag file will have 250 frames on the first run, then 252, then 255, 255, 253, 251, etc. I looked to see if it was simply dropping frames at the end, but the last frame was the same every time, so it is dropping somewhere else. Is this typical behavior? I am using `rs.config.enable_device_from_file(config, bag_fname, repeat_playback=False)` to finish the video and a `try` and `except` block to handle `frames = pipeline.wait_for_frames(timeout_ms=100)`.

As some context, I am concerned about this because I first get the 2D stream, then I do 2D pose estimation on that mp4 file, which returns 2D coordinates of body joints, and then I use the pyrealsense2 wrapper to align the color and depth frames to extract the 3D coordinate of those 2D locations. If the inconsistency is inherent, would it instead be recommended to align the depth and color streams and write them to video files before I do pose estimation, and then align those videos using opencv after?

Related issues: #4344

g2-bernotas, 2019-09-11:
Are you setting the playback not to play in real-time?

``` Python
    profile = pipe.start(config)
    playback=profile.get_device().as_playback()
    playback.set_real_time(False)
```

I have noticed before that I was loosing much more than just a few frames just because I was not running not in real-time.

kjacks21, 2019-09-11:
I'm streaming from a file though, which shouldn't have that issue (although that is a good point). Here is how I'm starting the stream:

```
config = rs.config()
rs.config.enable_device_from_file(config, bag_fname, repeat_playback=False)
pipeline = rs.pipeline()
config.enable_stream(rs.stream.color)
pipeline.start(config)
```

g2-bernotas, 2019-09-11:
Give it a try to include the set_real_time(False). Here is my setup that worked:

``` Python
    pipe = rs.pipeline()
    config = rs.config()

    rs.config.enable_device_from_file(config, bagfile)
    config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
    config.enable_stream(rs.stream.color, 640, 480, rs.format.rgb8, 30)
    profile = pipe.start(config)
    playback=profile.get_device().as_playback()
    playback.set_real_time(False)
```

kjacks21, 2019-09-11:
@bobdavies2000 That did the trick! Thank you!

closed 2019-09-11 by kjacks21