realsenseai/librealsense #4841: Intel Realsense viewer: Deepth stream start failure

Dontla, 2019-09-11:
[image]

Dontla, 2019-09-11:
And the next day I ran it in my nootbook, it worked well:
[image]
never wrong!
My nootbook information:
win10 64bit 2018 version, vs2015, cuda9.0 with the same cudnn, MX250 graphic card, Intel Realsense viewer2.25.0, camera D435, FW version05.11.06.250...

ev-mp, 2019-09-18:
@Dontla, I'm closing this now as there is no follow up to investigate.
Feel free to reopen if the issue resurfaces.

closed 2019-09-18 by ev-mp