realsenseai/librealsense #4845: C++ API  advanced mode JSON load doesn't actually apply settings

kyranf, 2019-09-11:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 05.10.03.00  | 
| Operating System & Version |   Win 10 | 
| Platform                                 | PC  |
| SDK Version                            |  {  2.20}                          | 
| Language                            |  { C++ }                          | 
| Segment			|  {Robot}                          | 

### Issue Description

Related to the question and comments/code examples shown in issue #4046 and also basically the exact same code/question/problems on this Stack Exchange question https://stackoverflow.com/questions/53520301/how-to-load-json-file-to-realsense-d435-camera-using-c-and-intel-api 

This is a snippet of what i'm doing, and i'll describe what happens and the relevant parts of the JSON. 

```
			rs400::advanced_mode advanced_mode_dev = dev->as<rs400::advanced_mode>();
			// Check if advanced-mode is enabled
			if (!advanced_mode_dev.is_enabled()) {
				// Enable advanced-mode
				advanced_mode_dev.toggle_advanced_mode(true);
				TB_LOG_INFO("Turned on advanced mode!");
			}
//read JSON from disk
			try {
				std::ifstream inputFile(JSONFilePath.c_str());
				std::string contents((std::istreambuf_iterator<char>(inputFile)), std::istreambuf_iterator<char>());
				advanced_mode_dev.load_json(contents);
			} catch (const rs2::error & e) {
				std::cerr << "RealSense error calling " << e.get_failed_function() << "(" << e.get_failed_args() << "):\n    " << e.what() << std::endl;
				TB_LOG_ERROR("REALSENSE ERROR - JSON file was NOT loaded successfully!");
			} catch (...) {
				TB_LOG_ERROR("OTHER ERROR - JSON file was NOT loaded successfully!");
			}

			TB_LOG_INFO("JSON file was loaded successfully!");
			STDepthTableControl depthTableGroup = advanced_mode_dev.get_depth_table(0); //get the struct full of current  values for depth control table settings
			depthTableGroup.disparityShift = 150;
			depthTableGroup.depthClampMin = 50;
			depthTableGroup.depthClampMax = 300;
			advanced_mode_dev.set_depth_table(depthTableGroup);

			depthTableGroup = advanced_mode_dev.get_depth_table(0); //get the struct full of current  values for depth control table settings
			TB_LOG_INFO("Confirming parameters, disparity shift: %d, min clamp %d, max clamp %d", depthTableGroup.disparityShift, depthTableGroup.depthClampMin, depthTableGroup.depthClampMax);
```

I can successfully save a JSON of the current camera's settings to disk, but i can't seem to successfully load one. It 'fails' silently, where the line `advanced_mode_dev.load_json(contents);` appears to execute successfully (with string contents which look good to me). If I then interrogate the depth control table where I have modified the disparity shift and min/max clamp distance values, they are NOT modified in the device after loading the JSON. I can, however, set the values myself in the struct and use the set_depth_table() function to set them. This is verified to work. 

But why won't the JSON load actually affect the camera, and not give any errors if it couldn't apply the settings? 

here's the first 17 lines of the JSON i'm loading which shows the relevant parameters which should be set when loaded:
```
{
    "aux-param-autoexposure-setpoint": "400",
    "aux-param-colorcorrection1": "0.461914",
    "aux-param-colorcorrection10": "-0.553711",
    "aux-param-colorcorrection11": "-0.553711",
    "aux-param-colorcorrection12": "0.0458984",
    "aux-param-colorcorrection2": "0.540039",
    "aux-param-colorcorrection3": "0.540039",
    "aux-param-colorcorrection4": "0.208008",
    "aux-param-colorcorrection5": "-0.332031",
    "aux-param-colorcorrection6": "-0.212891",
    "aux-param-colorcorrection7": "-0.212891",
    "aux-param-colorcorrection8": "0.68457",
    "aux-param-colorcorrection9": "0.930664",
    "aux-param-depthclampmax": "350",
    "aux-param-depthclampmin": "100",
    "aux-param-disparityshift": "150",
    "controls-autoexposure-auto": "False",
    "controls-autoexposure-manual": "8000",
```

So the depthClampMin should be 100 after I load the JSON and get the depth table. but it looks like it actually doesn't do anything. Here's a screenshot of the data after loading was 'successful', and i grab the devices' depth control table.
[image]

It only works properly after I manually set the parameters and write them back into the device with the set_depth_table() function as mentioned earlier. 

It would save me a lot of time if this worked, because if it doesn't work, i have to add all these parameters to my application's config structures and do a lot of extra boilerplate code to support this simple process with each parameter I need to use rather than just loading a neat JSON with all desired settings in one go.

dorodnic, 2019-09-11:
Settings loaded by load_json will be applied on next start. 
If I recall correctly, this is done because some of JSON supported values can be speculative dependent on the resolution, that we don't know ahead of time before start. 
When camera is streaming, load_json should behave as you'd expect. When the camera is not streaming, there is really no impact to any of these parameters.

kyranf, 2019-09-11:
I will test this situation, but before I was verifying/setting the depth control table, the actual streaming results didn't show the changes once the streams were started either.  So it might not show up due to what you are saying there, but ultimately it still failed to apply the changes when starting up.  But by setting them manually they DO work.

dorodnic, 2019-09-11:
I may be mistaken, but this is strange. 
I'm sure that setting value to `RS2_OPTION_VISUAL_PRESET` is delayed until start. 
What JSON are you using? Did you try the ones from [here](https://github.com/IntelRealSense/librealsense/wiki/D400-Series-Visual-Presets)? Does it behave the same when loading JSON from the Viewer?

RealSenseCustomerSupport, 2019-10-08:
----------------------------------------------

Hi @kyranf,

Do you try realsense-viewer to load the json file and see how it works?

RealSenseCustomerSupport, 2019-10-15:
----------------------------------------------

Hi @kyranf,

I don't observe this issue per my test with your preset on D415 + librealsense 2.28.0. If no other concern, I will close this ticket.

kyranf, 2019-10-15:
I will close it for now, until I have further tested.

closed 2019-10-15 by kyranf