realsenseai/librealsense #4846: Provide .deb source packages

santiagorr, 2019-09-11:
Please, consider publishing the sources of the librealsense .deb packages. [distribution_linux.md](https://github.com/IntelRealSense/librealsense/blob/e8cb6b4c051f57464ba8fdc334c65e04f5b57b90/doc/distribution_linux.md) only documents how to get binary deb packages (for ubuntu), but it would be great to have the sources.

Or even better, it would be great to have it directly in Debian (so in Ubuntu).

Cheers,
 -- S

ev-mp, 2019-09-18:
@santiagorr, thanks you for the inquiry .
Upstreaming the debians is planned activity and we're definitely looking into this. I cannot provide estimates though.
Releasing the sources is currently not planned, but once up-streamed, this may be reconsidered.

RealSenseCustomerSupport, 2019-10-07:
----------------------------------------------

Hi @santiagorr,

Do you need any further clarifications regarding this topic?

Thank you!

santiagorr, 2019-10-08:
> Hi @santiagorr,
> 
> Do you need any further clarifications regarding this topic?

Just that upstreaming in Debian "officially" (including the packages in debian.org, instead of/moreover having a private repository) would be more efficient for you in the long term, but that implies making the sources available.

thanks for you answer.

RealSenseSupport, 2020-06-05:
@santiagorr,

If that the only concern you have, I am going ahead and closing the issue, since we already have 
a thread where updates about the progress on "making the sources available" topic will be published: https://github.com/IntelRealSense/librealsense/issues/4431

Please don't hesitate to open a new thread if you have additional questions or concerns!
Thank you!

closed 2020-06-05 by RealSenseSupport