realsenseai/librealsense #4848: How to display Color, Depth, and Pose streams all at once?

brettmilinski, 2019-09-11:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 / T265 | 
| Firmware Version                   | 5.11.06.250 | 
| Operating System & Version |   Win (10)  | 
| Platform                                 | PC  |
| SDK Version                            |  2.25.0 | 
| Language                            | C# | 

### Issue Description
I'm trying to display all three of these streams simultaneously but I am crashing when I do so on the line:

`var pp = pipeline.Start(cfg); `

Is it not allowed to stream and display data from the D435 and T265 using the same pipeline?

```

try
     {
               Action<VideoFrame> updateDepth;
               Action<VideoFrame> updateColor;
               Action<VideoFrame> updatePose;
                // The colorizer processing block will be used to visualize the depth frames.
                colorizer = new Colorizer();

                // Create and config the pipeline to strem color and depth frames.
                pipeline = new Pipeline();

                var cfg = new Config();
                cfg.EnableStream(Stream.Depth, 640, 480);
                cfg.EnableStream(Stream.Color, Format.Rgb8);
                cfg.EnableStream(Stream.Pose);

                var pp = pipeline.Start(cfg);

                SetupWindow(pp, out updateDepth, out updateColor, out updatePose);
                Task.Factory.StartNew(() =>
                {
                    while (!tokenSource.Token.IsCancellationRequested)
                    {
                        // We wait for the next available FrameSet and using it as a releaser object that would track
                        // all newly allocated .NET frames, and ensure deterministic finalization
                        // at the end of scope. 
                        using (var frames = pipeline.WaitForFrames())
                        {
                            var colorFrame = frames.ColorFrame.DisposeWith(frames);
                            var depthFrame = frames.DepthFrame.DisposeWith(frames);
                            var poseFrame = frames.PoseFrame.DisposeWith(frames);

                            // We colorize the depth frame for visualization purposes
                            var colorizedDepth = colorizer.Process<VideoFrame>(depthFrame).DisposeWith(frames);

                            // Render the frames.
                            Dispatcher.Invoke(DispatcherPriority.Render, updateDepth, colorizedDepth);
                            Dispatcher.Invoke(DispatcherPriority.Render, updateColor, colorFrame);
                            Dispatcher.Invoke(DispatcherPriority.Render, updatePose, poseFrame);

                            Win32Point p = new Win32Point();
                            GetCursorPos(ref p);
                            //String distance = depthFrame.GetDistance(depthFrame.Width, depthFrame.Height).ToString();
                            //var vsp = pipeline.ActiveProfile.GetStream(Stream.Depth).As<VideoStreamProfile>();
                            // h = (p.X / (vsp.Height / 2));
                            // w = (p.Y / (vsp.Width / 2));
                            h = p.Y;
                            w = p.X;

                            Dispatcher.Invoke(new Action(() =>
                            {
                                // String depth_dev_sn = new Sensor.CameraInfos(depthFrame.Sensor)[CameraInfo.SerialNumber];
                                //String distance = (depthFrame.GetDistance(depthFrame.Width / 2, depthFrame.Height / 2)).ToString();
                                String distance = (depthFrame.GetDistance(w,h)).ToString();
                                String coordinates = "X: " + p.X.ToString() + "  Y: " + p.Y.ToString();

                                txtTimeStamp.Text = coordinates + " Distance: " + distance;
                            }));
                        } //end using
                    } //end while
                }, tokenSource.Token);  //end task && END OLD CODE KEEP AROUND*/
            }//end try
            catch (Exception ex)
            {
                MessageBox.Show(ex.Message);
                Application.Current.Shutdown();
            } //end catch

dorodnic, 2019-09-11:
Please take a look at [tracking-and-depth](https://github.com/IntelRealSense/librealsense/tree/master/examples/tracking-and-depth) example

brettmilinski, 2019-09-11:
@dorodnic Thank you I looked at those links and came up with the following for C#:

`
var dev = ctx.QueryDevices();
`
`
Config config = new Config();
`
`
//set all devices up
`
`
for(int i =0; i < dev.Count; i++)`
`
{
`
`
config.EnableDevice(dev[i].Info[CameraInfo.SerialNumber]);
`
`
config.EnableAllStreams();
`
`
Pipeline p = new Pipeline(ctx);
`
`
p.Start(config);
`
`
pipelines[i] = p;
`
`
}
`

However, I cannot seem to pick up the T265 device. When I debug, dev has a Count value of 1 so it's only retrieving the D435. Am I setting up the devices incorrectly here?

dorodnic, 2019-09-12:
Are you building the .NET wrapper from source or using the assembly file from the installer?

RealSenseCustomerSupport, 2019-10-22:
----------------------------------------------

@brettmilinski - Is this still happening?  As dorodnic asked is it consistently happening?  Were you able to get things working?

closed 2019-11-14 by RealSenseCustomerSupport