realsenseai/librealsense #4849: Camera Not being Detected In Linux

vidurvij, 2019-09-11:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Ubuntu 18   | 
| Kernel Version (Linux Only)    |  4.18.0-15-generic | 
| Platform                                 | PC |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C++ } |                          | 
| Segment			|                            | 

Camera is being detected and working on Windows 10 PC but it not being detected by the RealSense SDK. I think the OS detecting the camera as shown but not SDK.

[image]

Harish1901, 2019-09-11:
Hi, I am facing the same issue. I am able to compile but I do not see any output and also no error. How did you fix it.

ev-mp, 2019-09-18:
@vidurvij, did you manage to collect the logs ? 
Waiting for an update

vidurvij, 2019-09-18:
Hi Sorry for the late reply. The Problem was solved.

closed 2019-09-18 by vidurvij